nfcim/flutter_nfc_kit #3: PlatformException(406, No tag polled, null)

miroda, 2020-03-22:
```
[log] PlatformException(406, No tag polled, null)
[log] #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:569:7)
#1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:321:33)
```
请问这是这么回事？

Harry-Chen, 2020-03-22:
需要先调用 `poll` 寻卡后方可 `transceive`，具体请阅读 dart pub 上的文档

closed 2020-03-22 by Harry-Chen

barangungor, 2021-12-23:
I'm facing same error. What should i do?

dangfan, 2021-12-23:
You need to invoke `poll` first.
Please refer to the example at https://pub.dev/packages/flutter_nfc_kit

barangungor, 2021-12-23:
Thanks i'm getting this error

`Unhandled Exception: PlatformException(405, NDEF not supported on current tag, null, null)`

when using function in my page's InıtState like this:

```
FlutterNfcKit.poll(
            iosMultipleTagMessage: "Multiple tags found!",
            iosAlertMessage: "Scan your tag")
        .then((value) => FlutterNfcKit.readNDEFRecords(cached: false)
                .asStream()
                .listen((event) {
              print('event[0].identifier.toString()');
              print(event[0].idString.toString());
            }));
```

Harry-Chen, 2021-12-23:
@barangungor What type of tag & platform are you using?

jiegec, 2021-12-23:
@barangungor It seems that your tag does not support NDEF.

barangungor, 2021-12-23:
I'm using Android and my nfc card's and tag's types are NFC215 NTAG215. It doesn't work if my nfc's doesn't support NDEF?

dangfan, 2021-12-23:
Currently, we do not support formatting a tag. The tag can be used as an NDEF tag if and only if it is formatted to NDEF (with some metadata). You may use [NFC Tools](https://play.google.com/store/apps/details?id=com.wakdev.wdnfc) to format the tag first.

barangungor, 2021-12-23:
Ok thanks, do you support MIFARE Tag?