nfcim/flutter_nfc_kit #8: Crash when using iPhone 6

bottee, 2020-07-13:
Hi,

thank you for this great plug-in!

I have an issue when using this plug-in on an iPhone 6 without NFC support. I removed any call to the plug-in in this case, but I get following error during startup of my app:

dyld: Library not loaded: /usr/lib/swift/libswiftCoreNFC.dylib
  Referenced from: /private/var/containers/Bundle/Application/5A8733CD-0DD4-4356-9E37-C4567C2034A/Runner.app/Frameworks/flutter_nfc_kit.framework/flutter_nfc_kit
  Reason: image not found 

How to avoid this error?

jiegec, 2020-07-13:
What's your iOS version?

bottee, 2020-07-14:
On the iPhone 6 I have iOS 12.4.7 which is the last supported version.

After some more digging, I guess the plug-in should load the CoreNFC dylib only if needed. 

Maybe the podspec needs to configure it:

s.weak_frameworks = ['CoreNFC']

Example:

https://github.com/semlette/nfc_in_flutter/blob/master/ios/nfc_in_flutter.podspec

jiegec, 2020-07-14:
Nice catch, can you test whether it works? I don't have iOS12 devices at hand now.

bottee, 2020-10-01:
Sorry for my late response.

So I didn't got it working on iOS 12. I had to remove the plugin for old iOS versions.

hzjchina, 2020-10-23:
i got the sampe problem on iphone6  (12.4.8)
dyld: Library not loaded: /usr/lib/swift/libswiftCoreNFC.dylib
  Referenced from: /var/containers/Bundle/Application/2263992C-4487-4AB8-ADB5-97AE0E6D9064/Runner.app/Runner
  Reason: image not found
(lldb) 
libswiftCoreNFC must 13.0+ ?

jiegec, 2020-10-23:
> libswiftCoreNFC must 13.0+ ?

Yes, so how to handle this? I am not sure..

hzjchina, 2020-10-24:
I replaced with 'nfc_manager'. it is perfect for me

dangfan, 2021-04-02:
I’m closing this issue because libswiftCoreNFC requires iOS 13+.

Thank you!

commit b1c6ae5c (2020-08-12): Change metadata, try to fix #8
closed 2021-04-02 by dangfan