nfcim/flutter_nfc_kit #9: Change Reader Flags for Android Reader Mode

DanielPfannes, 2020-07-22:
Hi,

thank you for the great NFC plugin.

I would like to turn off default android sounds while reading a NFC tag. After some research I found out that this can be done by setting the flag FLAG_READER_NO_PLATFORM_SOUNDS (see [here](https://developer.android.com/reference/android/nfc/NfcAdapter#FLAG_READER_NO_PLATFORM_SOUNDS)), which is currently not set in your plugin. Is there a way to change the Reader flags in your plugin? It would be great to be able to do this with your plugin.

Harry-Chen, 2020-07-22:
It has been implemented now. However I will not publish it to `pub.dev` currently, you could use it directly in the form of remote git repo in your `pubspec.yaml`.

closed 2020-07-22 by Harry-Chen

DanielPfannes, 2020-07-22:
Thank you for the quick change! 
Unfortunately I'm getting a NullPointerException when polling for a tag with the updated version:

**Stack Trace**
[image]

**Dependency in pubspec.yaml**
[image]

**Executed Code**
`    await FlutterNfcKit.poll(
        timeout: Duration(hours: 2), 
        androidPlatformSound: false);`

Harry-Chen, 2020-07-22:
Sorry I use the wrong parameter name on Dart side. Now it should be fixed.

commit f19e51ac (2020-07-22): Support disabling platform sound on Android, fix #9