nfcim/flutter_nfc_kit #13: Failed to build iOS app

BryanBlack523, 2020-08-20:
Xcode: 11.6
Flutter: 1.17.5
Flutter_nfc_kit: 2.0.0

When launching sample app, build fails:
````
Xcode's output:
↳
    Command CompileSwift failed with a nonzero exit code
    Command CompileSwift failed with a nonzero exit code
    /Developer/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_nfc_kit-2.0.0/ios/Classes/SwiftFlutterNfcKitPlugin.swift:251:41: error: value of optional type 'NFCTypeNameFormat?' must be unwrapped to a value of type 'NFCTypeNameFormat'
                                    format: format,
                                            ^
    /Developer/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_nfc_kit-2.0.0/ios/Classes/SwiftFlutterNfcKitPlugin.swift:251:41: note: coalesce using '??' to provide a default when the optional value contains 'nil'
                                    format: format,
                                            ^
                                                   ?? <#default value#>
    /Developer/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_nfc_kit-2.0.0/ios/Classes/SwiftFlutterNfcKitPlugin.swift:251:41: note: force-unwrap using '!' to abort execution if the optional value contains 'nil'
                                    format: format,
                                            ^
                                                  !
    /Developer/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_nfc_kit-2.0.0/ios/Classes/SwiftFlutterNfcKitPlugin.swift:252:68: error: cannot convert value of type 'Any?' to expected argument type 'String'
                                    type: dataWithHexString(hex: record["type"]),
                                                                       ^
                                                                                as! String
    /Developer/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_nfc_kit-2.0.0/ios/Classes/SwiftFlutterNfcKitPlugin.swift:253:74: error: cannot convert value of type 'Any?' to expected argument type 'String'
                                    identifier: dataWithHexString(hex: record["identifier"]),
                                                                             ^
                                                                                            as! String
    /Developer/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_nfc_kit-2.0.0/ios/Classes/SwiftFlutterNfcKitPlugin.swift:254:71: error: cannot convert value of type 'Any?' to expected argument type 'String'
                                    payload: dataWithHexString(hex: record["payload"])
                                                                          ^
                                                                                      as! String
    Command CompileSwift failed with a nonzero exit code
    Command CompileSwift failed with a nonzero exit code
    note: Using new build system
    note: Building targets in parallel
    note: Planning build
    note: Constructing build description

Could not build the precompiled application for the device.

Error launching application on iPhone.
````

jiegec, 2020-08-20:
Fixed in #12

jiegec, 2020-08-20:
@Harry-Chen Please release a new version

closed 2020-08-20 by jiegec

Harry-Chen, 2020-08-21:
This problem is fixed in 2.0.1