nfcim/flutter_nfc_kit #15: Not needed entitlements

bottee, 2020-10-01:
Hello,

I found an issue starting since v1.2.0.

Since that time the iOS version tries to detect different types of NFC tags:

`NFCTagReaderSession(pollingOption: [.iso14443, .iso15693, .iso18092], delegate: self)`

before it only used 

`NFCTagReaderSession(pollingOption: [.iso14443], delegate: self)`

As i only use iso14443 tags in a security sensitive application, it is for me a security issue to add entitlements for unused tag technologies.

Would it be possible to add the tag type to FlutterNfcKit.poll ?

Thank you very much!

Harry-Chen, 2020-11-14:
This would be useful on both Android and iOS side to allow the caller to select only the NFC technologies they want. I currently do not have much time, but will work on it and include it in the next release.

Harry-Chen, 2020-11-15:
Fixed in 43c084e.

closed 2020-11-15 by Harry-Chen