nfcim/flutter_nfc_kit #16: Support EventChannel(s)

larssn, 2020-10-02:
Recommend supporting Streams through `EventChannel`, as it makes some operations easier (like having a listener running in the background).

Then it would be possible to having a simple API like:
```
var streamSub = FlutterNfcKit.tags.listen((tag) {
   print(tag);
});
```

Just a suggestion. Sadly I have no time to do a PR.

Harry-Chen, 2020-11-14:
This is possible, but will become a total different set of APIs. Maybe the foreground dispatch mode mentioned in #17 can work together with this Stream-like API.

luckycreationsindia, 2024-07-21:
How about this,

Stream<NFCTag> tags = Stream.fromFuture(poll(options));

var streamSub = tags.listen((tag) {
   print(tag);
});

luckycreationsindia, 2024-07-21:
I'll create pull request which will implement this feature and will not affect existing functionality.

Here's what will be changed:

New Class TagPollOptions will be Introduced:

`
class TagPollOptions {
  TagPollOptions({
    this.timeout,
    this.androidPlatformSound = true,
    this.androidCheckNDEF = true,
    this.iosAlertMessage = "Hold your iPhone near the card",
    this.iosMultipleTagMessage = "More than one tags are detected, please leave only one tag and try again.",
    this.readIso14443A = true,
    this.readIso14443B = true,
    this.readIso18092 = false,
    this.readIso15693 = true,
    this.probeWebUSBMagic = false,
  });

  final Duration? timeout;
  final bool androidPlatformSound;
  final bool androidCheckNDEF;
  final String iosAlertMessage;
  final String iosMultipleTagMessage;
  final bool readIso14443A;
  final bool readIso14443B;
  final bool readIso18092;
  final bool readIso15693;
  final bool probeWebUSBMagic;
}
`

New method to set options for stream:

`FlutterNFCKit.setTagStreamOptions = TagPollOptions(...)`

New stream:

`
var streamSub = FlutterNFCKit.tags.listen((tag) {
   print(tag);
});
`

Harry-Chen, 2024-07-21:
@luckycreationsindia It is great if you could submit a PR on EventChannel support. Looking forward to it!

luckycreationsindia, 2024-07-21:
> @luckycreationsindia It is great if you could submit a PR on EventChannel support. Looking forward to it!

It won't be EventChannel but layer in flutter itself. Let me paste the line of code before committing.

`
static TagPollOptions? _options;

  static set setTagStreamOptions(TagPollOptions? options) {
    _options = options;
  }

  static Stream<NFCTag> tags = Stream.fromFuture(poll(
    timeout: _options?.timeout,
    androidPlatformSound: _options?.androidPlatformSound ?? true,
    androidCheckNDEF: _options?.androidCheckNDEF ?? true,
    iosAlertMessage: _options?.iosAlertMessage ?? "Hold your iPhone near the card",
    iosMultipleTagMessage:
        _options?.iosMultipleTagMessage ?? "More than one tags are detected, please leave only one tag and try again.",
    readIso14443A: _options?.readIso14443A ?? true,
    readIso14443B: _options?.readIso14443B ?? true,
    readIso18092: _options?.readIso18092 ?? false,
    readIso15693: _options?.readIso15693 ?? true,
    probeWebUSBMagic: _options?.probeWebUSBMagic ?? false,
  ));
`

luckycreationsindia, 2024-07-22:
I've created EventChannel Support for streaming and pull request is added for the same with other information.

Note:

- I've to change sourceCompatibility to JavaVersion.VERSION_1_8 and targetCompatibility to JavaVersion.VERSION_1_8 with kotlin jvmTarget to '1.8' due to incompatibility and will ensure latest flutter version support
- Example is also added for streaming tag
- **Streaming will only support Android as of now. I don't know how to handle this in iOS and Web**

commit 601f3032 (2024-12-21): fix #16, close #179
closed 2024-12-21 by Harry-Chen