nfcim/flutter_nfc_kit #20: NDEF not supported on current tag

wrjcs, 2020-11-26:
您好,我的卡是mifare classic,我用您例子里面的代码读取数据,就直接报错了
if (tag.type == NFCTagType.mifare_classic) {
            //print(tag.toJson());
            var ndefRecords = await FlutterNfcKit.readNDEFRecords();
            //print(jsonEncode(ndefRecords).toString());
            var ndefString = ndefRecords.map((r) => r.toString()).reduce((value, element) => value + "\n" + element);
            //print(ndefString);
          }

[ERROR:flutter/lib/ui/ui_dart_state.cc(177)] Unhandled Exception: PlatformException(405, NDEF not supported on current tag, null, null)
E/flutter (25307): #0      StandardMethodCodec.decodeEnvelope
package:flutter/…/services/message_codecs.dart:582
E/flutter (25307): #1      MethodChannel._invokeMethod
package:flutter/…/services/platform_channel.dart:159
E/flutter (25307): <asynchronous suspension>
E/flutter (25307): #2      MethodChannel.invokeMethod
package:flutter/…/services/platform_channel.dart:332
E/flutter (25307): #3      FlutterNfcKit.readNDEFRawRecords
package:flutter_nfc_kit/flutter_nfc_kit.dart:249
E/flutter (25307): #4      FlutterNfcKit.readNDEFRecords
package:flutter_nfc_kit/flutter_nfc_kit.dart:236

Harry-Chen, 2020-11-26:
此问题是因为卡片支持的 technology 中不包含 ndef 数据，见 https://github.com/nfcim/flutter_nfc_kit/blob/eae6aa7ba897122751ebe3eec16761e7a9b508c4/android/src/main/kotlin/im/nfc/flutter_nfc_kit/FlutterNfcKitPlugin.kt#L379-L387

我们的示例仅适用于卡片已经被 NDEF 格式化的情形

wrjcs, 2020-11-26:
您好,那我要怎样才能读取我这个卡片上的数据呢?
谢谢

dangfan, 2020-11-27:
从目前我们看到的错误信息来看，你是使用Android手机从M1（Mifare Classic）卡里读取NDEF，而这种卡并不包含NDEF数据。我的建议是，请先使用NXP TagInfo之类的工具验证它确实包含你想获取的数据，然后再来考虑读写方法。

dangfan, 2021-04-02:
I’m closing this issue because it has been inactive for a few months.

Please reopen if you still encounter this issue.

Thank you!

closed 2021-04-02 by dangfan