nfcim/flutter_nfc_kit #21: App crash when run on ios 14.1

saobang187, 2020-11-27:
Hi Sir,
I followed the setup in pub.dev for ios(add to Info.plist file), but the app crash when going to the widget using FlutterNfcKit.poll or FlutterNfcKit.transceive, then NFCAvailability.available work ok only. I don't know how to fix this issue, many thank if you can help me resolve it.
Thanks,
The Anh

dangfan, 2020-11-28:
Do you have any related logs?

saobang187, 2020-11-28:
Yes sir, all config file and debug log after disconnect device.
[image]
[image]
[image]
[image]

saobang187, 2020-11-28:
@dangfan maybe channel name error because I change channel name by other string then app don't crash, but it don't send data to card until.
[image]

dangfan, 2020-11-29:
@saobang187 From the information you provided, we couldn't figure out what happened. Would you please provide a minimal reproducible example? Thanks!

saobang187, 2020-11-29:
@dangfan i use your code in pub.dev(you can see in images), and call cardConnect method(use FlutterNfcKit.poll with parameters for ios) in the initState of a widget(you can create any widget and add this), app crash when go to line FlutterNfcKit.poll, i tested on iphone7 plus and xs.

saobang187, 2020-11-30:
@dangfan apple support Read NFC for smartMx3 NXP, i try NFC other tag then it work ok, so I will close this issue at here. Thanks you

closed 2020-11-30 by saobang187