nfcim/flutter_nfc_kit #22: PlatformException(500, Invalidate session with error, Missing required entitlement, null) on iOS 14.1.

saobang187, 2020-12-01:
Hi sir,
I followed your configuration in Infor.plist file and try run your example on iOS 14.1, I click polling start button then get the message: PlatformException(500, Invalidate session with error, Missing required entitlement, null).
Please help me check this issue.
Thanks you!

jiegec, 2020-12-01:
Have you configured NFC entitlements?

saobang187, 2020-12-01:
@jiegec yep, i configured in Runner.entitlements file below:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>com.apple.developer.nfc.readersession.formats</key>
	<array>
		<string>NDEF</string>
		<string>TAG</string>
	</array>
</dict>
</plist>

saobang187, 2020-12-01:
@jiegec Hi sir,
add nfc in capabilities Xcode and this issue is resolved but polling is not return value. Follow your example code I have:
var tag = await FlutterNfcKit.poll(
        timeout: Duration(seconds: 60),
        iosMultipleTagMessage: "Multiple tags found!",
        iosAlertMessage: "Scan your tag");
print(jsonEncode(tag));

But tag wait forever and don't print to console. Can you help me check it?

closed 2020-12-04 by saobang187

devnullpointer, 2021-03-24:
@saobang187  I'm also not getting the nfc dialog for start scan. I get a platform exception in debug mode, what did you do to solve as you closed this ticket?