nfcim/flutter_nfc_kit #24: popup waiting to scan on android

rezashamdani, 2021-01-07:
hi,

first i want to say thank you for your times to create this plugin.

is it possible to have the same alert popup waiting to scan on android phone?

big thanks

dangfan, 2021-01-07:
Thank you for the advice.
Unfortunately, we have no such plan recently. As you may see in #16, the typical behavior on Android is background dispatching. So in the near future, we might implement this feature first.

Harry-Chen, 2021-01-07:
Also, the popup windows on iOS is a system functionality (that means when programmers can not control its behaviour), but Android does not provide similar API. 

Ou project [nfsee](https://github.com/nfcim/nfsee) has implemented a popup window on Android. You may want to refer to code in its repo.

rezashamdani, 2021-01-07:
thank you for your quick response.

i hope it will be implemented soon.

thanks again.

closed 2021-01-07 by rezashamdani