nfcim/flutter_nfc_kit #31: Error running on iOS target

kasvith, 2021-04-21:
I've set up the entitlements as listed in the readme. But when try to run on iOS I get the following Pod error

```
    /Library/Ruby/Gems/2.6.0/gems/cocoapods-1.10.1/lib/cocoapods/installer.rb:239:in `block in resolve_dependencies'
    /Library/Ruby/Gems/2.6.0/gems/cocoapods-1.10.1/lib/cocoapods/user_interface.rb:64:in `section'
    /Library/Ruby/Gems/2.6.0/gems/cocoapods-1.10.1/lib/cocoapods/installer.rb:238:in `resolve_dependencies'
    /Library/Ruby/Gems/2.6.0/gems/cocoapods-1.10.1/lib/cocoapods/installer.rb:160:in `install!'
    /Library/Ruby/Gems/2.6.0/gems/cocoapods-1.10.1/lib/cocoapods/command/install.rb:52:in `run'
    /Library/Ruby/Gems/2.6.0/gems/claide-1.0.3/lib/claide/command.rb:334:in `run'
    /Library/Ruby/Gems/2.6.0/gems/cocoapods-1.10.1/lib/cocoapods/command.rb:52:in `run'
    /Library/Ruby/Gems/2.6.0/gems/cocoapods-1.10.1/bin/pod:55:in `<top (required)>'
    /usr/local/bin/pod:23:in `load'
    /usr/local/bin/pod:23:in `<main>'
Error running pod install
Error launching application on iPhone 11 Pro Max.
Exited (sigterm)
```

kasvith, 2021-04-21:
Fixed it by setting min iOS level in Podfile to '13'

closed 2021-04-21 by kasvith

martin-braun, 2025-09-09:
But in the (hopefully running) example it's not locked: https://github.com/nfcim/flutter_nfc_kit/blob/c452afab81d1881963c42714b352594c4f803ccf/example/ios/Podfile#L2