nfcim/flutter_nfc_kit #32: iOS iso7816 transceive response error

whywilson, 2021-04-29:
I'm working on reading an iso1443-4 card on iOS. It works well on Swift code but failed in the flutter project.

## Code with flutter_nfc_kit
```
var tag = await FlutterNfcKit.poll(timeout: Duration(seconds: 10), iosMultipleTagMessage: "Multiple tags found!", iosAlertMessage: "Scan your tag");
    print(tag.type);

    if (tag.type == NFCTagType.iso7816) {
      var result = await FlutterNfcKit.transceive(Uint8List.fromList([0x00, 0xA4, 0x04, 0x00, 0x08, 0x50, 0x41, 0x59, 0x2E, 0x54, 0x49, 0x43, 0x4C, 0x00]));
      print(result);
    }
```
## flutter_nfc_kit Response
```
flutter: FlutterNfcKit poll
flutter: NFCTagType.iso7816
flutter: [106, 130]
[CoreNFC] 00000002 8001c870 -[NFCTagReaderSession setAlertMessage:]:90  (null)
[CoreNFC] 00000002 8001c870 -[NFCTagReaderSession setAlertMessage:]:90  (null)
```

The command is correct, it works well in Xcode with swift.
## Code with Swift
```
let selectApp : NFCISO7816APDU = NFCISO7816APDU(data: Data.init([0x00, 0xA4, 0x04, 0x00, 0x08, 0x50, 0x41, 0x59, 0x2E, 0x54, 0x49, 0x43, 0x4C, 0x00]))!
            
session.connect(to: tag0!) { (e: Error?) in tag?.sendCommand(apdu: selectApp, completionHandler: { (data: Data, sw1: UInt8, sw2: UInt8, error: Error?) in
  print("selectApp APDU Result \(data.hexDescription)")
})
}
```
## Swift Response
```
selectApp APDU Result 6f3484085041592e5449434ca5289f0801029f0c21ffffffffffffffff000000000000000000000000000000002019082400000186a0
```

Please help to confirm the issues. 🙏

jiegec, 2021-04-29:
Hi, did you add NDEF AID to Info.plist?

Harry-Chen, 2021-04-29:
`106, 130` is 0x6A. 0x82, which means `NDEF Tag Application not found`, which might result from missing AID in App configuration.

whywilson, 2021-04-29:
The AID is already added to Info.plist. If not, this ISO1443 Card won't be detected at all.

Harry-Chen, 2021-04-29:
> The AID is already added to Info.plist. If not, this ISO1443 Card won't be detected at all.

Is it `D2760000850101`?

whywilson, 2021-04-29:
> > The AID is already added to Info.plist. If not, this ISO1443 Card won't be detected at all.
> 
> Is it `D2760000850101`?

Nope, it's 5041592E5449434C and 5041592E41505059 for transport card [羊城通].
I guess the **tag** used to **connect** was not converted to **ISO7816TAG**. 
Here's the full swift code which works well. 
```
        func tagReaderSession(_ session: NFCTagReaderSession, didDetect tags: [NFCTag]) {
            print("didDetect \(tags)")
            var nfcIso7816Tag: NFCISO7816Tag? = nil
            var nfcTag: NFCTag? = nil
            
            for tag in tags {
                if case let .iso7816(cTag) = tag {
                    nfcIso7816Tag = cTag
                    nfcTag = tag
                }
            }
            
            if nfcIso7816Tag == nil {
                session.invalidate(errorMessage: "Card not support")
            }
            
            let selectApp : NFCISO7816APDU = NFCISO7816APDU(data: Data.init([0x00, 0xA4, 0x04, 0x00, 0x08, 0x50, 0x41, 0x59, 0x2E, 0x54, 0x49, 0x43, 0x4C, 0x00]))!
            let readPurse : NFCISO7816APDU = NFCISO7816APDU(data: Data.init([0x80, 0x5C, 0x00, 0x02, 0x04]))!
                                   ⬇️
            session.connect(to: nfcTag!) { (e: Error?) in
                         ⬇️
                nfcIso7816Tag?.sendCommand(apdu: selectApp, completionHandler: { (data: Data, sw1: UInt8, sw2: UInt8, error: Error?) in
                    print("selectApp APDU Result \(data.hexDescription)")
                })
                
                nfcIso7816Tag?.sendCommand(apdu: readPurse, completionHandler: { (data: Data, sw1: UInt8, sw2: UInt8, error: Error?) in
                    let moneyString  = data.hexDescription
                    let moneyValue = Double(Int(moneyString, radix: 16)!) / 100.0
                    print("Data \(data.hexDescription)")
                    session.alertMessage = "羊城通 余额：￥\(moneyValue)"
                })
            }
        }
```

Harry-Chen, 2021-04-29:
It is cast to `iso7816` indeed.

https://github.com/nfcim/flutter_nfc_kit/blob/a7e277ae4cb6ae8b492ad693607d5acb55d3245c/ios/Classes/SwiftFlutterNfcKitPlugin.swift#L95-L117

Harry-Chen, 2021-04-29:
We will try to reproduce and investigate this.

jiegec, 2021-04-30:
Sorry, but I could not reproduce the problem:

```
                        Uint8List result1 =
                            await FlutterNfcKit.transceive(Uint8List.fromList([
                          0x00,
                          0xA4,
                          0x04,
                          0x00,
                          0x08,
                          0x50,
                          0x41,
                          0x59,
                          0x2E,
                          0x54,
                          0x49,
                          0x43,
                          0x4C,
                          0x00
                        ]));
                        setState(() {
                          _result = '1: $result1\n';
                        });
```

It returns an array like [111, 52, 132, ...].

whywilson, 2021-05-05:
> Sorry, but I could not reproduce the problem:
> 
> ```
>                         Uint8List result1 =
>                             await FlutterNfcKit.transceive(Uint8List.fromList([
>                           0x00,
>                           0xA4,
>                           0x04,
>                           0x00,
>                           0x08,
>                           0x50,
>                           0x41,
>                           0x59,
>                           0x2E,
>                           0x54,
>                           0x49,
>                           0x43,
>                           0x4C,
>                           0x00
>                         ]));
>                         setState(() {
>                           _result = '1: $result1\n';
>                         });
> ```
> 
> It returns an array like [111, 52, 132, ...]. I am using 羊城通.

Thank you for the testing. But it's still no normal response here.
Here's the step of how I use the repository. 

1. var tag = await FlutterNfcKit.poll(xxxxxx)
2. check tag type if it's NFCTagType.iso7816.
3. var result = await FlutterNfcKit.transceive
4. print the result map and always got "6A 82".

Would you please check or show the key code of your example?

jiegec, 2021-05-05:
> Would you please check or show the key code of your example?

I used the example project in the repo. Only two modifications are done:

1. Add the aid to info.plist
2. Change user logic to the code aforementioned.

whywilson, 2021-05-06:
> > Would you please check or show the key code of your example?
> 
> I used the example project in the repo. Only two modifications are done:
> 
> 1. Add the aid to info.plist
> 2. Change user logic to the code aforementioned.

Just tried on different cards.
The code works well on the virtual card in the OnePlus phone, but not on a real card. 
This has been tested on Android and iOS devices.
Still checking if the batch of the card matters. Maybe the one I tested is too old.(P2012-15120)

dangfan, 2021-05-06:
I suggest you use a contactless card reader (e.g., ACR122u) to test the card first.

whywilson, 2021-05-06:
> I suggest you use a contactless card reader (e.g., ACR122u) to test the card first.

It still gets a 6A 82 response with the PN532 reader. Maybe it's the issues with the real card.
[image]

If the latest card works, this can be confirmed exactly.

dangfan, 2021-05-06:
It seems that the old 羊城通 cannot be select using AID. Instead, you could try to use the short DF ID (e.g., DDF1 and ADF1) to select the application. Unfortunately, this is impossible on iOS due to the limitation of CoreNFC.

closed 2021-05-06 by dangfan

whywilson, 2021-05-06:
> It seems that the old 羊城通 cannot be select using AID. Instead, you could try to use the short DF ID (e.g., DDF1 and ADF1) to select the application. Unfortunately, this is impossible on iOS due to the limitation of CoreNFC.

But both cards and virtual cards work well on SwiftUI with CoreNFC with the same code which lists here. Any other suggestions?
https://github.com/nfcim/flutter_nfc_kit/issues/32#issuecomment-828922125

dangfan, 2021-05-06:
So you mean that the card which returns 6A82 when using PN532 works fine with the code you provided?
If so, I suggest that you use a reader to test the selection order:
In Lingnan Pass, there are two applications with different AIDs (see https://wiki.nfc.im/books/%E6%99%BA%E8%83%BD%E5%8D%A1%E6%89%8B%E5%86%8C/page/%E5%B2%AD%E5%8D%97%E9%80%9A%EF%BC%88lingnan-pass%EF%BC%89).
Please try to select the first AID 5041592E41505059  and then select the second AID 5041592E5449434C and then swap the order to see if there are any differences.