nfcim/flutter_nfc_kit #37: iso7816 support in android

Abdktefane, 2021-07-27:
I am not an expert in NFC, but I want to ask can I connect android nfc card with specifc AID with APDU (iso7816)
i can see that ios have aid field in info.plist
Is it supported in Android without specifying it?

jiegec, 2021-07-27:
I think that you do not need to write an allowlist of AID in Android.

Harry-Chen, 2021-07-27:
What card are you using and what error / exception did you encounter? Can you post minimal code for reproduction?

smlu, 2021-08-09:
@Abdktefane reading the ISO 7816 icc should work on android without a problem. You don't need to specify the AID as you have to for iOS.

Fyi, I've made [dmrtd](https://github.com/ZeroPass/dmrtd) library for reading biometric passports using this library.
For testing, you can try [ICC](https://github.com/ZeroPass/dmrtd/blob/df707067bd7afb021441a35cfb8ec70fbc88434c/lib/src/proto/iso7816/icc.dart#L25-L244) class and read any ISO 7816-4 supported card that has NFC connection interface.