nfcim/flutter_nfc_kit #40: NFC not working on Android 10, 11

jafar260698, 2021-09-15:
It is not detecting anything on Symsung  A51, A71. what should i do, are there anything special for them ?

Harry-Chen, 2021-09-15:
Sorry but we do not have Samsung devices. Do you mean you can see NFC enabled but cannot poll tags?

jafar260698, 2021-09-15:
Yeah , it is not polling anythig, also it is not throwing anthing

Harry-Chen, 2021-09-15:
This is very strange, can you see anything unnormal in Flutter debug log? Are you sure you add the technologies (such as NFCA, MirareClassic` to your manifest? Can other applications read cards (such as TagInfo by NXP)?

jafar260698, 2021-09-15:
It is giving such error 

-------------++ PlatformException(500, Communication error, Tag was lost., null) #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:597:7)
I/flutter ( 7187): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:158:18)
I/flutter ( 7187): <asynchronous suspension>
I/flutter ( 7187): #2      FlutterNfcKit.transceive (package:flutter_nfc_kit/flutter_nfc_kit.dart:225:12)
I/flutter ( 7187): <asynchronous suspension>
I/flutter ( 7187): #3      NFCCardChannel.transmit (package:Eimzo/provider/nfc_provider.dart:16:21)
I/flutter ( 7187): <asynchronous suspension>

Harry-Chen, 2021-09-24:
Sorry for the late reply, please check:

* What option are you using for `poll`?
* Can other applications, such as TagInfo, poll tags normally?

closed 2021-12-06 by dangfan

hebaothman, 2024-06-02:
What's the solution please?

Harry-Chen, 2024-06-12:
We have no Samsung devices, and we are not able to reproduce the problem on other devices.

Should anyone answer the question in https://github.com/nfcim/flutter_nfc_kit/issues/40#issuecomment-926642090, we might be able to proceed with the problem.

kauemurakami, 2025-03-11:
2025 with xiaomi redmi note 8 the same error when write, any solutions?