nfcim/flutter_nfc_kit #41: Password Protection

domgamba, 2021-09-20:
Any consideration for functionality creating password encryption with NFC tags? I have NTAG213 and I need to allow my users to protect their tag so that nobody else can overwrite their content.

Harry-Chen, 2021-09-24:
Sure you can implement it with the `transceive` API we provide by using appropriate APDUs. We actually *plan* to add more specific support for different types of tags/cards. However it is (obviously) time-consuming and we cannot provide any guarantee about this functionality.

alaashahin1, 2021-12-28:
@Harry-Chen Can you help me how use transceive method

jiegec, 2021-12-28:
> @Harry-Chen Can you help me how use transceive method

Please read the docstring.

alaashahin1, 2021-12-28:
@jiegec I read Docs but I don't know What data are to be sent to nfcv.transceive(......);
to unlock tag then write on tag.

jiegec, 2021-12-28:
> @jiegec I read Docs but I don't know What data are to be sent to nfcv.transceive(......); to unlock tag then write on tag.

You should refer to the docs of the nfc tag.

dangfan, 2021-12-28:
@alaashahin1 Please follow this documentation: https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf
To authenticate a tag, you need to use the PWD_AUTH command (in Section 10.7). This command is what you need to transceive.

alaashahin1, 2021-12-28:
@dangfan @jiegec  thanks all for consideration.
I read this Section 10.7, but frankly, I don't know what i will do in flutter

sachindivergents, 2023-08-28:
can you guide me how to how i can fix the issue?