nfcim/flutter_nfc_kit #42: Reader Mode Flags (FLAG_READER_NO_PLATFORM_SOUNDS) and (FLAG_READER_SKIP_NDEF_CHECK) are misconfigured

KhatibFX, 2021-09-23:
Currently, in the library code:

```dart
if (!androidCheckNDEF) technologies |= 0x100;
if (!androidPlatformSound) technologies |= 0x80;
```

but, according to the android documentation:

```txt
FLAG_READER_NO_PLATFORM_SOUNDS

Setting this flag allows the caller to prevent the platform from playing sounds when it discovers a tag.

Constant Value: 256 (0x00000100)
----------------------------------------------------------------------------------------
FLAG_READER_SKIP_NDEF_CHECK

Setting this flag allows the caller to prevent the platform from performing an NDEF check on the tags it finds.

Constant Value: 128 (0x00000080) 
```

I think you misconfigured the flags so that the checkNDEF flag actually manipulates the androidPlatformSound, and the platformSound actually manipulates the checkNDEF flag. I verified this issue by setting the platform sound to false and therefore, being unable to read NDEF tags.

Also, thanks for this library!

Harry-Chen, 2021-09-24:
Sorry for this, and I will release a fixed version soon.

Harry-Chen, 2021-09-24:
Released in v3.0.2.

closed 2021-09-24 by Harry-Chen