nfcim/flutter_nfc_kit #43: NFCTag class isn't null-safe

rostopira, 2021-09-29:
I've tried to read tag with this library on iOS, got 
```
Unhandled Exception: type 'Null' is not a subtype of type 'String' in type cast
#0      _$NFCTagFromJson (package:flutter_nfc_kit/flutter_nfc_kit.g.dart:14:18)
#1      new NFCTag.fromJson (package:flutter_nfc_kit/flutter_nfc_kit.dart:108:57)
#2      FlutterNfcKit.poll (package:flutter_nfc_kit/flutter_nfc_kit.dart:215:19)
<asynchronous suspension>
#3      NfcPageState.readFromTag (package:profit_tracker/nfc_page.dart:29:17)
<asynchronous suspension>
```
NFCTag has every field as non-null, however even docs say the opposite:
```
  /// ATQA (Type A only, Android only)
  final String atqa;
```

Harry-Chen, 2021-09-29:
Thank you for reporting! We will fix it soon.

germancvdev, 2021-10-13:
Please could you correct this problem? I have a project that depends on this package and when updating all to **_Null safety_** the message happens.