nfcim/flutter_nfc_kit #53: Support make the NDEF tag readonly

timnew, 2021-12-03:
I searched the API list but could not find any way to make the Ndef tag readonly.

Would be nice to support following code:

```dart
if (await FlutterNfcKit.isWritable()) {
  await FlutterNfcKit.makeReadOnly();
}
```

For Android `Ndef.isWritable`, `Nedf.canMakeReadOnly` and `Nedf.makeReadOnly` does the trick.

For iOS, `NFCNDEFTag.queryNEDFStatus` and NFCNDEFTag.writeLock` does the trick.

I'm a bit busy for now, might be able to provide PR later.

timnew, 2021-12-06:
@Harry-Chen would you mind to have a look at PR https://github.com/nfcim/flutter_nfc_kit/issues/53

I haven't been able to test the code on my local, as there are some files are missing from the repo.

Harry-Chen, 2021-12-06:
> @Harry-Chen would you mind to have a look at PR #53

Sure! Thanks for your PR.

> I haven't been able to test the code on my local, as there are some files are missing from the repo.

Could you describe this problem with more detail? I could compile the project (freshly cloned from git) without error.

Harry-Chen, 2021-12-06:
Fixed in e7cfebc along with #54.

closed 2021-12-06 by Harry-Chen

timnew, 2021-12-07:
Ahh, I see `3.2.0`