nfcim/flutter_nfc_kit #55: androidx.core:core-ktx:1.7 fails release build

timnew, 2021-12-13:
The library `androidx.core:core-ktx:1.7` now causes resource linking issue and seems no easy fix for now except downgrade.

```
* What went wrong:
Execution failed for task ':flutter_nfc_kit:verifyReleaseResources'.
> A failure occurred while executing com.android.build.gradle.tasks.VerifyLibraryResourcesTask$Action
   > Android resource linking failed
     ERROR:/Users/timnew/.gradle/caches/transforms-3/fa2e903d500609c2badd44e01c2b21b1/transformed/core-1.7.0/res/values/values.xml:105:5-114:25: AAPT: error: resource android:attr/lStar not found.
```

Check more:
* https://stackoverflow.com/questions/69033022/message-error-resource-androidattr-lstar-not-found
* https://stackoverflow.com/questions/69041630/android-build-error-lstar-not-found
* https://stackoverflow.com/questions/69021225/resource-linking-fails-on-lstar

Harry-Chen, 2021-12-18:
Thanks! I will check on this later

Harry-Chen, 2021-12-18:
Would it help to increase `compileSdkVersion` to `31` in `android/build.gradle`? Seems this is a more clean fix than forcibly downgrading `core-ktx`.

lore-co, 2021-12-23:
> Would it help to increase `compileSdkVersion` to `31` in `android/build.gradle`? Seems this is a more clean fix than forcibly downgrading `core-ktx`.

I've a similar error:

`The class is loaded from /Users/lorenzocotti/.gradle/caches/transforms-2/files-2.1/6dc09ef492479f456878d1ba74749970/jetified-kotlin-stdlib-1.6.0.jar!/kotlin/Unit.class
e: /Users/lorenzocotti/Documents/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_nfc_kit-3.2.0/android/src/main/kotlin/im/nfc/flutter_nfc_kit/FlutterNfcKitPlugin.kt: (249, 32): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.6.0, expected version is 1.1.15.`

Changing both the targetSdk and compileSdk to 31 doesn't work.

Harry-Chen, 2021-12-23:
@lore-co The error you encountered seems not related to this issue. You might need to upgrade to the latest version of Kotlin (`>=1.6.0`) in your project.

timnew, 2021-12-23:
I'll have a try to upgrade the compile SDK version after the holiday.
I thought I might have tried it,or I thought it is upgrading the minimal SDK version.
Anyway, I'll have a look and post the result here

timnew, 2021-12-23:
And merry Xmas 🎄

kelegorm, 2022-01-12:
> @lore-co The error you encountered seems not related to this issue. You might need to upgrade to the latest version of Kotlin (>=1.6.0) in your project.

That helped me! Thank you!

timnew, 2022-01-27:
Not sure anyone find a solution of this bug other than downgrade `androidx.core:core-ktx` to `1.6.0`.

I have kind of upgraded everything to bleeding edge, but still get the error:
```
classpath 'com.android.tools.build:gradle:7.1.0'
classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:1.6.10"
classpath 'com.google.gms:google-services:4.3.10'
classpath 'com.google.firebase:firebase-crashlytics-gradle:2.8.1'

compileSdkVersion 32
minSdkVersion 21
targetSdkVersion 32

implementation "org.jetbrains.kotlin:kotlin-stdlib-jdk8:1.6.10"
```

timnew, 2022-01-28:
I kind of managed to fixed the issue by adding
```
implementation 'androidx.appcompat:appcompat:1.4.1'
implementation 'androidx.appcompat:appcompat-resources:1.4.1'
```

kelegorm, 2022-02-02:
BTW, I fixed it in new way. My old didn't work after all. So, new solution is:
Stick to new kotlyn and upgrade gradle. What I have in first `build.gradle`:
```
buildscript {
    ext.kotlin_version = '1.6.0'

    repositories {
        google()
        mavenCentral()
    }

    dependencies {
        classpath 'com.android.tools.build:gradle:4.1.0'
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:$kotlin_version"
    }
}
```

In the `app/build.gradle` I use compileSdkVersion 31, minSdkVersion 21, targetSdkVersion 30.

In gradle/gradle-wrapper.properties I have:
`distributionUrl=https\://services.gradle.org/distributions/gradle-6.7-all.zip`

Also I use flutter 2.5.3 (didn't upgrade to 2.8 yet), flutter_nfc_kit: ^3.1.0. And this way I can build app on codemagic.

Also to make building works locally I have upgraded kotlyn and gradle plugins for AndroidStudio. Now it's 203-1.6.10 and gradle is 203.7717.56, bundled to my Android Studio 2020.3.1 patch 3.

I hope that will help you too!

JAICHANGPARK, 2022-02-28:
I had same problem ~!

closed 2022-05-11 by Harry-Chen

Harry-Chen, 2022-05-11:
By bumping `compileSdkVersion` to `31` in the plugin, I believe this problem can be solved.

commit cf9a3926 (2022-05-11): Bump compileSdkVersion to fix #55