nfcim/flutter_nfc_kit #61: NFC Tag not reading and throwing weird exception

migalv, 2022-01-05:
I keep getting this error in my Firebase Crashlytics console for Android users. It's not letting them scan their NFC tags properly.

Does someone know why this is happening?

This seems to happen on some phones. The model is not the issue because it works for Samsung S1 model and also throws this exception for other users with the same device...

Exception:
```
PlatformException(error, context cannot be null, null, java.lang.IllegalArgumentException: context cannot be null
	at android.nfc.NfcAdapter.getDefaultAdapter(NfcAdapter.java:871)
	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin.handleMethodCall(FlutterNfcKitPlugin.kt:63)
	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin.onMethodCall(FlutterNfcKitPlugin.kt:58)
	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:233)
	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:85)
	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:818)
	at android.os.MessageQueue.nativePollOnce(Native Method)
	at android.os.MessageQueue.next(MessageQueue.java:336)
	at android.os.Looper.loop(Looper.java:197)
	at android.app.ActivityThread.main(ActivityThread.java:8167)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)
). Error thrown NFC Unknown Error.
       at StandardMethodCodec.decodeEnvelope(message_codecs.dart:581)
       at MethodChannel._invokeMethod(platform_channel.dart:158)
       at FlutterNfcKit.poll(flutter_nfc_kit.dart:204)
```

Flutter doctor:
```
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel unknown, 2.0.3, on macOS 11.4 20F71 darwin-arm, locale en-GB)
[✓] Android toolchain - develop for Android devices (Android SDK version 32.0.0)
[✓] Xcode - develop for iOS and macOS
[✓] Chrome - develop for the web
[✓] Android Studio (version 2020.3)
[✓] VS Code (version 1.63.2)
[✓] Connected device (2 available)

• No issues found!
```

Flutter version: v.2.0.3

Package version: v2.2.1

dangfan, 2022-01-06:
Could you add a breakpoint at `onAttachedToActivity` (https://github.com/nfcim/flutter_nfc_kit/blob/master/android/src/main/kotlin/im/nfc/flutter_nfc_kit/FlutterNfcKitPlugin.kt#L284) to check if `binding.activity` is null?

migalv, 2022-01-06:
Hello @dangfan, thanks for your reply.

Could you please tell me how can I do that? I've never set a breakpoint in an Android file while working with Flutter. I use VSCode.

Also, I'm receiving this error from my users. I haven't been able to reproduce it on my end.

Harry-Chen, 2022-05-11:
Seems `activity` is null at:

https://github.com/nfcim/flutter_nfc_kit/blob/2546b962a320b095e827348f93da7a2bdcd6bb57/android/src/main/kotlin/im/nfc/flutter_nfc_kit/FlutterNfcKitPlugin.kt#L64

This is weird (just like #71). As @dangfan mentioned, `activity` should be obtained in `onAttachedToActivity`. Anyway, I will add a guard there to check nullity.

Harry-Chen, 2022-05-11:
I cannot reproduce this. Maybe this is related to some strange behaviour on modified Android systems. I will close this issue. If you can find a reproducer, feel free to re-open it.

commit b4d99f38 (2022-05-11): Add more null check in Android plugin, fix #61 and #71
closed 2022-05-11 by Harry-Chen