nfcim/flutter_nfc_kit #62: iOS Background Reading

thomasklaush, 2022-01-18:
Is it possible to do iOS Background reading?

Meaning reading a tag without the annoying pop-up of the iPhone.

mekjolle, 2022-01-19:
I was about to ask this question myself. 

Would be a really nice feature and should be possible with the CORENFC library having added this feature lately. 
Please refer to the following link -> [Adding Support for Background Tag Reading](https://developer.apple.com/documentation/corenfc/adding_support_for_background_tag_reading)

timnew, 2022-01-20:
Based on the code 
```swift
func application(_ application: UIApplication,
                 continue userActivity: NSUserActivity,
                 restorationHandler: @escaping ([Any]?) -> Void) -> Bool {

    guard userActivity.activityType == NSUserActivityTypeBrowsingWeb else {
        return false
    }

    // Confirm that the NSUserActivity object contains a valid NDEF message.
    let ndefMessage = userActivity.ndefMessagePayload
    guard ndefMessage.records.count > 0,
        ndefMessage.records[0].typeNameFormat != .empty else {
            return false
    }

    // Send the message to `MessagesTableViewController` for processing.
    guard let navigationController = window?.rootViewController as? UINavigationController else {
        return false
    }

    navigationController.popToRootViewController(animated: true)
    let messageTableViewController = navigationController.topViewController as? MessagesTableViewController
    messageTableViewController?.addMessage(fromUserActivity: ndefMessage)

    return true
}
```
I think it can't be fully handled by a FlutterPlugin. Instead you might need to go to the Runner app to read the tag info and do the deep linking yourself as any other universal/deeplinking related scenarios

Harry-Chen, 2022-05-11:
It's something similar to Android's foreground dispatching (#17). In these modes, you do not have to actively call some API to poll a tag. The OS will call your handler instead.

However as a Flutter plugin, it is difficult to act as a handler. You **have** to add some rather complex native code to your project. So we might not be able to support this easily.

spehj, 2023-09-04:
I was wondering if anyone has successfully implemented iOS background reading in Flutter. Doing this requires some native code, and it's challenging for me to find any useful resources.

Has anyone progressed on this issue? If so, could you possibly share some directions or even link to a GitHub repository with a Flutter example?

Your insights would be incredibly helpful for those of us looking to implement this feature.

Thank you!

plut9, 2023-09-05:
> I was wondering if anyone has successfully implemented iOS background reading in Flutter. Doing this requires some native code, and it's challenging for me to find any useful resources.
>
> Has anyone [...]

I'd be interested in this as well as we're trying to achieve background NFC scanning on iOS for a customer-specific feature.

zigapovhe, 2023-09-07:
Need this too! I'm very surprised this doesn't exist yet.

HenriqueMatias, 2024-04-01:
Hi everyone. Any development on this item?

spehj, 2024-04-02:
> Hi everyone. Any development on this item?

I had to implement it on my own. I wrote some iOS native code and used method channels.

HenriqueMatias, 2024-04-07:
Hey Spehj thank you for the hint.
So did you fork the project or you wrote the code from scratch?

spehj, 2024-04-07:
@HenriqueMatias I wrote it from scratch in AppDelegate.swift