nfcim/flutter_nfc_kit #68: Feature request : ISO 15693 commands for Android and iOS

nicolasgarnet, 2022-03-01:
Hi,

Since iOS opened custom [commands to ISO 15693](https://developer.apple.com/documentation/corenfc/nfciso15693tag), it would be nice to have sending ISO 15693 tags commands available for Android and iOS.

(Maybe this is already possible with ISO 7816, but I am not familiar with that standard and I don't know if it includes ISO 15693 interactions)

Harry-Chen, 2022-05-11:
https://developer.apple.com/documentation/corenfc/nfciso15693tag/3043799-customcommand
https://developer.android.com/reference/android/nfc/tech/NfcV#transceive(byte[])

Android and iOS provide different interfaces on ISO15693. We might need to wrap around this.

Harry-Chen, 2022-05-15:
@nicolasgarnet We do not have ISO 15693 tags currently. If you do, could your help with testing on:

* iOS document for `customCommand` mentions `Sends a custom command (0xA0 to 0xDF command code)`, does it support other commands (that are already wrapped by other methods such as `readSingleBlock`?
* How does `customCommand` on iOS report errors returned by card? Is it a `ISO15693TagResponseErrorCode` in `NSError` returned?
* How does `transceive` of `NfcV` on Android report errors? I suppose it would be encoded in the returned byte array.

Thanks!

khanliKU, 2022-08-01:
Command : Get Sys Info Command 20 2B E0 2B 00 60 00 00 71 5B

When I try to run transceiver command on a ISO15693 tag it works on Android and throws the error:

PlatformException(405, Transceive not supported on this type of card, null, null)

Does this have anything to do with this issue?

I am stuck on iOS please help :)

Harry-Chen, 2022-08-03:
> Command : Get Sys Info Command 20 2B E0 2B 00 60 00 00 71 5B
>
> When I try to run transceiver command on a ISO15693 tag it works on Android and throws the error:
>
> PlatformException(405, Transceive not [...]

@khanliKU Yes, it is due to that we do not currently have ISO 15693 support on iOS.

Harry-Chen, 2023-10-31:
I investigated again and found that seems <https://developer.apple.com/documentation/corenfc/nfciso15693tag/3585168-sendrequest> can be used to send arbitrary commands.

Harry-Chen, 2023-10-31:
Implemented in https://github.com/nfcim/flutter_nfc_kit/commit/1388333f130f80bf02fd91615004e812ed0963ad

closed 2023-10-31 by Harry-Chen