nfcim/flutter_nfc_kit #71: Null pointer exception

rohnsht, 2022-04-06:
I getting this crashes registered in Firebase crashlytics.

`Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'android.view.Window android.app.Activity.getWindow()' on a null object reference
       at android.nfc.NfcActivityManager$NfcActivityState.<init>(NfcActivityManager.java:138)
       at android.nfc.NfcActivityManager.getActivityState(NfcActivityManager.java:202)
       at android.nfc.NfcActivityManager.disableReaderMode(NfcActivityManager.java:252)
       at android.nfc.NfcAdapter.disableReaderMode(NfcAdapter.java:2023)
       at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$pollTag$$inlined$schedule$1.run(FlutterNfcKitPlugin.java:6)
       at java.util.TimerThread.mainLoop(TimerThread.java:562)
       at java.util.TimerThread.run(TimerThread.java:512)`

rohnsht, 2022-05-03:
I figured out the issue. The app was crashing because the timer tried to disable nfc when the app is in background. I just added a null check for activity before disabling nfcReaderMode.

Harry-Chen, 2022-05-11:
😕  This is not expected, because `pollingTimeoutTask` should be cancelled once the plugin is detached from activity.

https://github.com/nfcim/flutter_nfc_kit/blob/6daf38e6664ab312bd1bf772e1796f34aebf5565/android/src/main/kotlin/im/nfc/flutter_nfc_kit/FlutterNfcKitPlugin.kt#L291-L297

rohnsht, 2022-05-11:
> 😕 This is not expected, because `pollingTimeoutTask` should be cancelled once the plugin is detached from activity.
> 
> https://github.com/nfcim/flutter_nfc_kit/blob/6daf38e6664ab312bd1bf772e1796f34aebf5565/android/src/main/kotlin/im/nfc/flutter_nfc_kit/FlutterNfcKitPlugin.kt#L291-L297

I don't this **inDetachedFromActivity()** gets called when the app is in background.

closed 2022-05-11 by Harry-Chen