nfcim/flutter_nfc_kit #72: Execution failed for task ':flutter_nfc_kit:compileDebugKotlin'.

sudhanshupearlorganisation, 2022-04-06:
e: C:\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_nfc_kit-3.2.0\android\src\main\kotlin\im\nfc\flutter_nfc_kit\FlutterNfcKitPlugin.kt: (429, 13): Class 'MethodResultWrapper' is not abstract and does not implement abstract member public abstract fun error(p0: String, p1: String?, p2: Any?): Unit defined in io.flutter.plugin.common.MethodChannel.Result
e: C:\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_nfc_kit-3.2.0\android\src\main\kotlin\im\nfc\flutter_nfc_kit\FlutterNfcKitPlugin.kt: (449, 9): 'error' overrides nothing
e: C:\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_nfc_kit-3.2.0\android\src\main\kotlin\im\nfc\flutter_nfc_kit\FlutterNfcKitPlugin.kt: (452, 40): Type mismatch: inferred type is String? but String was expected

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_nfc_kit:compileDebugKotlin'.
> Compilation error. See log for more details

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 36s
Exception: Gradle task assembleDebug failed with exit code 1

Foldblade, 2022-04-19:
Same issues here, with:
```
Flutter 2.13.0-0.1.pre • channel beta • https://github.com/flutter/flutter.git
Framework • revision 13a2fb10b8 (7 days ago) • 2022-04-12 15:34:25 -0500
Engine • revision 499984f99c
Tools • Dart 2.17.0 (build 2.17.0-266.1.beta) • DevTools 2.12.1
```

larssn, 2022-05-03:
Had the same problem with a beta version of Flutter. Not expecting it to be fixed in stable, but it will turn into a problem when the current beta version of Flutter goes stable.

Harry-Chen, 2022-05-11:
Thanks! We will fix it in the next release

commit e2993b89 (2022-05-11): Change function signature of MethodResultWrapper, fix #72