nfcim/flutter_nfc_kit #73: Flutter 3 error

Mythar, 2022-05-15:
e: C:\Flutter 3\.pub-cache\hosted\pub.dartlang.org\flutter_nfc_kit-3.2.0\android\src\main\kotlin\im\nfc\flutter_nfc_kit\FlutterNfcKitPlugin.kt: (429, 13): Class 'MethodResultWrapper' is not abstract and does not implement abstract member public abstract fun error(p0: String, p1: String?, p2: Any?): Unit defined in io.flutter.plugin.common.MethodChannel.Result
e: C:\Flutter 3\.pub-cache\hosted\pub.dartlang.org\flutter_nfc_kit-3.2.0\android\src\main\kotlin\im\nfc\flutter_nfc_kit\FlutterNfcKitPlugin.kt: (449, 9): 'error' overrides nothing
e: C:\Flutter 3\.pub-cache\hosted\pub.dartlang.org\flutter_nfc_kit-3.2.0\android\src\main\kotlin\im\nfc\flutter_nfc_kit\FlutterNfcKitPlugin.kt: (452, 40): Type mismatch: inferred type is String? but String was expected

Harry-Chen, 2022-05-15:
Duplicated with #73. Will be solved in the next release.

samueldelamotte, 2022-05-16:
@Harry-Chen When do you think that you'll be publishing the next release?

Harry-Chen, 2022-05-17:
See v3.3.0.

closed 2022-05-17 by Harry-Chen