nfcim/flutter_nfc_kit #77: Miissing entitlement

farawehassan, 2022-06-13:
I keep getting this error while polling 
[CoreNFC] 00000002 82c5c2d0 -[NFCTagReaderSession _connectTag:error:]:686  Error Domain=NFCError Code=2 "Missing required entitlement" UserInfo={NSLocalizedDescription=Missing required entitlement}

This is my info.plist and Runner.entitlements file
[image]

[image]

jiegec, 2022-06-13:
Missing entitlement checklist:

- Add [Near Field Communication Tag Reader Session Formats Entitlements](https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_developer_nfc_readersession_formats) to your entitlements.
- Add [NFCReaderUsageDescription](https://developer.apple.com/documentation/bundleresources/information_property_list/nfcreaderusagedescription) to your Info.plist.
- Add [com.apple.developer.nfc.readersession.felica.systemcodes](https://developer.apple.com/documentation/bundleresources/information_property_list/systemcodes) and [com.apple.developer.nfc.readersession.iso7816.select-identifiers](https://developer.apple.com/documentation/bundleresources/information_property_list/select-identifiers) to your Info.plist as needed. WARNING: for iOS 14.5 and earlier versions, you MUST add them before invoking poll with readIso18092 or readIso15693 enabled, or your NFC WILL BE TOTALLY UNAVAILABLE BEFORE REBOOT due to a https://github.com/nfcim/flutter_nfc_kit/issues/23.
- Open Runner.xcworkspace with Xcode and navigate to project settings then the tab Signing & Capabilities.
- Select the Runner in targets in left sidebar then press the "+ Capability" in the left upper corner and choose Near Field Communication Tag Reading.

farawehassan, 2022-06-13:
> Missing entitlement checklist:
> 
> * Add [Near Field Communication Tag Reader Session Formats Entitlements](https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_developer_nfc_readersession_formats) to your entitlements.
> * Add [NFCReaderUsageDescription](https://developer.apple.com/documentation/bundleresources/information_property_list/nfcreaderusagedescription) to your Info.plist.
> * Add [com.apple.developer.nfc.readersession.felica.systemcodes](https://developer.apple.com/documentation/bundleresources/information_property_list/systemcodes) and [com.apple.developer.nfc.readersession.iso7816.select-identifiers](https://developer.apple.com/documentation/bundleresources/information_property_list/select-identifiers) to your Info.plist as needed. WARNING: for iOS 14.5 and earlier versions, you MUST add them before invoking poll with readIso18092 or readIso15693 enabled, or your NFC WILL BE TOTALLY UNAVAILABLE BEFORE REBOOT due to a [Ios 14.2(iPhone XR) Dialog NFC scan don't show #23](https://github.com/nfcim/flutter_nfc_kit/issues/23).
> * Open Runner.xcworkspace with Xcode and navigate to project settings then the tab Signing & Capabilities.
> * Select the Runner in targets in left sidebar then press the "+ Capability" in the left upper corner and choose Near Field Communication Tag Reading.

I did all of this which is in the screenshot I shared above but I still get the error

vaimikpatel2908, 2022-07-29:
I am heaving the same error. I tried Nfc_Manager's latest version as well. Both failed on iOS with the same error. It throws an error "Missing required entitlement" while calling the .poll method.

I have verified the following things:
I have enabled the capability for the app identifier in the apple developer account.
Also ,Followed the check list which is provided in the ReadMe section of the plugin to add Entitlement and NFCReaderUsageDescription in info.plist.

Does anyone has figured out why it is not working yet?

SimonHausdorf, 2022-11-08:
Running into the same issue. Followed checklist, I also tried the example in the repository which should work without issues?

dangfan, 2022-11-09:
Would you please provide a minimal reproducible example?

Tabarek, 2023-05-16:
any update on this?

wonSllaB, 2023-05-30:
> any update on this?

On iOS, payment AIDs are currently not allowed to be polled.
- AID 'A000000003...' is the standard first 5 bytes of a VISA payment application AID
- If you try this with another AID (non-payment) you would be able to start the polling and transceive
- **Note that this restriction also applies to select commands send through FlutterNfcKit.tranceive()**
  - A similar error would be prompted by CoreNFC when you try to do so

closed 2023-09-06 by Harry-Chen

aliangsama, 2024-04-11:
I also encountered the same problem

> I am heaving the same error. I tried Nfc_Manager's latest version as well. Both failed on iOS with the same error. It throws an error "Missing required entitlement" while calling the .poll method. [...]