nfcim/flutter_nfc_kit #88: build apk file failed

smurat, 2022-11-02:
when i try 'flutter build apk' or 'flutter build --release' with connected device on example app, this error occured

ERROR:C:\Users\smcelik\.gradle\caches\transforms-3\8f284dc91965176e19c58dd82841e0ed\transformed\jetified-kotlin-reflect-1.6.21.jar: R8: Type kotlin.reflect.jvm.internal.impl.serialization.deserialization.builtins.BuiltInsResourceLoader is defined multiple times: C:\Users\smcelik\.gradle\caches\transforms-3\8f284dc91965176e19c58dd82841e0ed\transformed\jetified-kotlin-reflect-1.6.21.jar:kotlin/reflect/jvm/internal/impl/serialization/deserialization/builtins/BuiltInsResourceLoader.class, C:\Users\smcelik\.gradle\caches\transforms-3\8f284dc91965176e19c58dd82841e0ed\transformed\jetified-kotlin-reflect-1.6.21.jar:META-INF/versions/9/kotlin/reflect/jvm/internal/impl/serialization/deserialization/builtins/BuiltInsResourceLoader.class

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:minifyReleaseWithR8'.
> com.android.tools.r8.CompilationFailedException: Compilation failed to complete, origin: C:\Users\smcelik\.gradle\caches\transforms-3\8f284dc91965176e19c58dd82841e0ed\transformed\jetified-kotlin-reflect-1.6.21.jar:kotlin/reflect/jvm/internal/impl/serialization/deserialization/builtins/BuiltInsResourceLoader.class

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BU�LD FAILED in 1m 46s
Running Gradle task 'assembleRelease'...                          107,5s

┌─ Flutter Fix ──────────────────────────────────────────────────────────────┐
│ [!] The shrinker may have failed to optimize the Java bytecode.            │
│ To disable the shrinker, pass the `--no-shrink` flag to this command.      │
│ To learn more, see: https://developer.android.com/studio/build/shrink-code │
└────────────────────────────────────────────────────────────────────────────┘
Gradle task assembleRelease failed with exit code 1

bayramaliunuvar, 2023-06-25:
Does anyone have any information regarding the problem?

huseyinbedir, 2023-09-05:
Is there any progress on this issue?

Harry-Chen, 2023-09-06:
I will look into this problem today. Sorry for the delay.

huseyinbedir, 2023-09-06:
> I will look into this problem today. Sorry for the delay.

Thank you. This issue is very important to me. i will be waiting 🙃

closed 2023-09-06 by Harry-Chen

bayramaliunuvar, 2023-11-20:
I'm still getting the same error.

**Error**
`* What went wrong:
Execution failed for task ':app:minifyReleaseWithR8'.
> com.android.tools.r8.CompilationFailedException: Compilation failed to complete, origin: C:\Users\BAU\.gradle\caches\transforms-3\1d004bcf53e866d0bf97c62d164a539c\transformed\jetified-kotlin-reflect-1.9.10.jar:kotlin/reflect/jvm/internal/impl/serialization/deserialization/builtins/BuiltInsResourceLoader.class`

**Flutter doctor**
`[√] Flutter (Channel stable, 3.13.5, on Microsoft Windows [Version 10.0.22621.2715], locale tr-TR)
    • Flutter version 3.13.5 on channel stable at C:\src\flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 12fccda598 (9 weeks ago), 2023-09-19 13:56:11 -0700
    • Engine revision bd986c5ed2
    • Dart version 3.1.2
    • DevTools version 2.25.0

[√] Windows Version (Installed version of Windows is version 10 or higher)

[√] Android toolchain - develop for Android devices (Android SDK version 33.0.2)
    • Android SDK at C:\Users\BAU\AppData\Local\Android\Sdk
    • Platform android-34, build-tools 33.0.2
    • Java binary at: C:\Program Files\Android\Android Studio\jbr\bin\java
    • Java version OpenJDK Runtime Environment (build 17.0.6+0-b2043.56-9586694)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[√] Visual Studio - develop Windows apps (Visual Studio Community 2022 17.7.6)
    • Visual Studio at C:\Program Files\Microsoft Visual Studio\2022\Community
    • Visual Studio Community 2022 version 17.7.34221.43
    • Windows 10 SDK version 10.0.22000.0

[√] Android Studio (version 2022.2)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 17.0.6+0-b2043.56-9586694)

[√] VS Code (version 1.79.1)
    • VS Code at C:\Users\BAU\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.76.0

[√] Connected device (3 available)
    • Windows (desktop) • windows • windows-x64    • Microsoft Windows [Version 10.0.22621.2715]
    • Chrome (web)      • chrome  • web-javascript • Google Chrome 119.0.6045.160
    • Edge (web)        • edge    • web-javascript • Microsoft Edge 119.0.2151.72

[√] Network resources
    • All expected network resources are available.`

**pubspec.yaml**
`dependencies:
  flutter:
    sdk: flutter
  flutter_localizations:
    sdk: flutter

  # The following adds the Cupertino Icons font to your application.
  # Use with the CupertinoIcons class for iOS style icons.
  cupertino_icons: ^1.0.5
  font_awesome_flutter: ^10.3.0
  material_design_icons_flutter: ^7.0.7296
  http: ^1.0.0
  connectivity: ^3.0.6
  shared_preferences: ^2.0.17
  url_launcher: ^6.1.8
  # fluttertoast: 8.2.4
[... 128 lines cut ...]
    id 'com.android.application' version '7.0.4' apply false
    id 'com.android.library' version '7.0.4' apply false
    id 'org.jetbrains.kotlin.android' version "$kotlin_version" apply false
}

allprojects {
    repositories {
        gradlePluginPortal()
        google()
        mavenCentral()
    }
}

rootProject.buildDir = '../build'
subprojects {
    project.buildDir = "${rootProject.buildDir}/${project.name}"
}
subprojects {
    project.evaluationDependsOn(':app')
}

tasks.register("clean", Delete) {
    delete rootProject.buildDir
}
`

commit fbeae129 (2023-09-06): Fix compilation, bump dependencies, fix #88