nfcim/flutter_nfc_kit #89: how to get uri

k3v1n-uncle, 2023-01-10:
for (var record in await FlutterNfcKit.readNDEFRecords(cached: false)) {
debugPrint(record.toString());
}
the result: [UriRecord: id=(empty) typeNameFormat=TypeNameFormat.nfcWellKnown type=U uri=https://github.com/nfcim/flutter_nfc_kit]
now, I want get the "uri=https://github.com/nfcim/flutter_nfc_kit" for string type, But I do not know how to do! please help me! thanks

sant0s, 2023-04-29:
@k3v1n-uncle record.payload will give you the payload as a string in hexadecimal:

- the first byte i.e. the first 2 hexadecimal characters represents the [URI prefix](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/nfc/NdefRecord.java#241). for example, `01` represents `http://www.`
- the remaining bytes represent the rest of the URI

Harry-Chen, 2023-09-06:
@sant0s Thanks for your reply!

closed 2023-09-06 by Harry-Chen