nfcim/flutter_nfc_kit #90: app kill due to nfc plugin

thomasklaush, 2023-01-13:
App dies due to NFC Plugin
Android 13 
Pixel6 Pro

E/AndroidRuntime(26060): FATAL EXCEPTION: Thread-97
E/AndroidRuntime(26060): Process: com.smaspos.kubikos, PID: 26060
E/AndroidRuntime(26060): java.lang.SecurityException: Permission Denial: Tag ( ID: 04 3D 54 AA 53 60 80 ) is out of date
E/AndroidRuntime(26060): 	at android.nfc.Tag.getTagService(Tag.java:381)
E/AndroidRuntime(26060): 	at android.nfc.tech.BasicTagTechnology.isConnected(BasicTagTechnology.java:63)
E/AndroidRuntime(26060): 	at android.nfc.tech.Ndef.isConnected(Ndef.java:71)
E/AndroidRuntime(26060): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$2.invoke(FlutterNfcKitPlugin.kt:125)
E/AndroidRuntime(26060): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$2.invoke(FlutterNfcKitPlugin.kt:118)
E/AndroidRuntime(26060): 	at kotlin.concurrent.ThreadsKt$thread$thread$1.run(Thread.kt:30)

rohnsht, 2023-01-17:
I have got the same issue with my app. The app crashes after reading a nfc  tag.

alperu, 2023-05-16:
I think answer for this might be here.

https://stackoverflow.com/questions/75695662/securityexception-while-ndef-connect-on-android-13

UPDATE: Thanks to your feedback and some additional research I came to a solution.

I was triggering this NFC write from the onNewIntent function. It seems that it is no longer possible due to some security issues. I solved it by triggering this NFC write from the onResume function instead.

Zardozz, 2023-06-02:
The fix for this is in https://github.com/nfcim/flutter_nfc_kit/blob/master/android/src/main/kotlin/im/nfc/flutter_nfc_kit/FlutterNfcKitPlugin.kt where you capture exceptions like `catch (ex: IOException)` from Tag operations that generate exceptions to either catch the generic exception `catch (ex: Exception)` or capture the specific new `SecurityException`

Harry-Chen, 2023-09-06:
Thanks @alperu @Zardozz for your help! It seems that using stored `Tag` after the physical tag is removed would trigger this Exception. At this time I decide not to change the logic in Android native code, but to capture `SecurityException` and gracefully returns error to Flutter side.

commit 99aa6d48 (2023-09-06): Capture SecurityException in Android code, fix #90
closed 2023-09-06 by Harry-Chen