nfcim/flutter_nfc_kit #94: Getting 409 when trying to poll a tag

PoeMcMoe, 2023-01-30:
Tested on the example app, iOS 15.4.1, iPhone 12, plugin ver 3.3.1

1. Press start polling.
2. Scan a tag.
3. When the 'Finished' message appears dismiss it by swiping down.
4. Press start polling again immediately (bug does not appear if you wait a couple of seconds).
5. You will get the 'Session invalidated by user' exception and the 'Ready to scan message' will still be displayed. The user will not be able to scan a tag even though there is a prompt to do so.

mofukommit, 2023-02-10:
Having mostly the same error / exception.
When starting the poll, even if I closed the existing poll with 
`await FlutterNfcKit.finish(iosErrorMessage: "Failed",);`
the error appears if I want to scan again. 
It is difficult to check if the session is closed or not. Or why the session is 'Session invalidated'?

sant0s, 2023-07-14:
according to the [documentation](https://github.com/nfcim/flutter_nfc_kit), `finish()` must be called once only. are you sure that 1) `finish()` is called once only AND 2) only after a successful tag scan? if a tag scan is not successful, then a session is not started. in this case, `finish()` would not need to be called.

Harry-Chen, 2023-09-06:
Closing due to no recent activity. Feel free to re-open anytime.

closed 2023-09-06 by Harry-Chen