nfcim/flutter_nfc_kit #104: iOS overlay removal

sant0s, 2023-06-01:
during tag scanning on iOS, there's an overlay that includes a top line of text like "Ready to read", an animated image of a device moving and a bottom line of text like "Hold your iPhone near the card".

is there a way to remove this overlay altogether?

jiegec, 2023-06-02:
I don't think so, this popup is shown by iOS.

sant0s, 2023-06-17:
thanks. while some sources indicate that's indeed [not possible](https://stackoverflow.com/questions/47708668/how-to-remove-popup-message-of-nfc-tag-in-ios), others hint [otherwise](https://stackoverflow.com/questions/50464810/is-it-possible-to-hide-ios-system-alert-for-nfc-reading-session).

jiegec, 2023-06-17:
> thanks. while some sources indicate that's indeed [not possible](https://stackoverflow.com/questions/47708668/how-to-remove-popup-message-of-nfc-tag-in-ios), others hint [otherwise](https://stackoverflow.com/questions/50464810/is-it-possible-to-hide-ios-system-alert-for-nfc-reading-session).

The post suggests that the popup can be hidden automatically **after** scanning the card, but not before.

sant0s, 2023-06-28:
That's right. Closing this issue.

closed 2023-06-28 by sant0s