nfcim/flutter_nfc_kit #109: i have checked the latest version of the flutter_nfc_kit and the nfc manager as well and i am pretty much sure that there session are not getting collapsed so still that issue which you are suggesting is releted to session collapse? #

Rahul123723, 2023-09-06:
i have checked the latest version of the flutter_nfc_kit and the nfc manager as well and i am pretty much sure that there session are not getting collapsed so still that issue which you are suggesting is releted to session collapse? #

_Originally posted by @Rahul123723 in https://github.com/nfcim/flutter_nfc_kit/issues/108#issuecomment-1708284178_

Harry-Chen, 2023-09-06:
I cannot understand your problem. Please DO NOT spam our issues by posting duplicated content everywhere.

closed 2023-09-06 by Harry-Chen

Rahul123723, 2023-09-06:
I didn't understand what you are talking about spam and duplicate content I posted my problem on github issues which I had faced how did I spammed your issue if you are not able to help me don't close my issue and continue with your stuff thank you.