nfcim/flutter_nfc_kit #112: Unable to read/write tags

suraj740, 2023-09-12:
Hi, i am using mifare ultralight, ndef tag on my android phone (Android Version 12) using the example application provided. 
Its unable to read/write tag using the example application.

Other apps such as NFC tool is able to read/write the same tag.

Harry-Chen, 2023-09-13:
Please provide some code to reproduce the problem, and any error log when using `flutter_nfc_kit`.

suraj740, 2023-09-13:
I have NFC tag which has some data in it, but the example application is unable to read or write which is provided by flutter_nfc_kit.

I am running the same example which is present flutter_nfc_kit repo under example folder.

Harry-Chen, 2023-09-14:
> I have NFC tag which has some data in it, but the example application is unable to read or write which is provided by flutter_nfc_kit.
> 
> I am running the same example which is present flutter_nfc_kit repo under example folder.

Are you using the write ndef functionality? Please provide some logs so we can look into this problem.

suraj740, 2023-09-14:
Below are the logs which i am getting while writing,

```
,0-1080,2295 #1 aid=1073741824} flags=0 reason=SHOW_SOFT_INPUT
D/InsetsController( 9057): show(ime(), fromIme=true)
D/InputMethodManager( 9057): showSoftInput() view=io.flutter.embedding.android.FlutterView{2774a4e VFE...... .F...... 0,0-1080,2295 #1 aid=1073741824} flags=0 reason=SHOW_SOFT_INPUT
D/InsetsController( 9057): show(ime(), fromIme=true)
D/InputMethodManager( 9057): showSoftInput() view=io.flutter.embedding.android.FlutterView{2774a4e VFE...... .F...... 0,0-1080,2295 #1 aid=1073741824} flags=0 reason=SHOW_SOFT_INPUT
D/InsetsController( 9057): show(ime(), fromIme=true)
I/flutter ( 9057): #0      _$NFCTagFromJson (package:flutter_nfc_kit/flutter_nfc_kit.g.dart:47:52)
I/flutter ( 9057): #1      new NFCTag.fromJson (package:flutter_nfc_kit/flutter_nfc_kit.dart:144:57)
I/flutter ( 9057): #2      FlutterNfcKit.poll (package:flutter_nfc_kit/flutter_nfc_kit.dart:261:19)
I/flutter ( 9057): <asynchronous suspension>
I/flutter ( 9057): #3      _MyAppState.build.<anonymous closure> (package:flutter_nfc_kit_example/main.dart:172:44)
I/flutter ( 9057): <asynchronous suspension>
```

and while reading when i click in start polling , nothing happens 
no logs are coming
[image]

Harry-Chen, 2023-09-15:
Seems no tags are polled. Maybe you can just invoke `poll` and see if it returns.

suraj740, 2023-09-15:
Even after invoking poll no logs are printing after this,
```
 NFCTag tag = await FlutterNfcKit.poll();
 print('tag'); //  this is not printing in console
```

Harry-Chen, 2023-10-30:
Sorry but with no further details, we cannot help to diagnose your problem

amasuke55, 2023-11-22:
Hello, I am experiencing the same issue.
I am using a Samsung S20FE with Android 13.
The tag is an ISO 14443-3A NXP - Mifare Classic 1k

I am just trying to read the tag, because I need the serialnumber from it.
I am using the provided example code in this repo and the latest nfc kit version as of writing this.
After I turned on all exceptions this is the exception I get.
[image]

Charly6596, 2023-11-28:
> Hello, I am experiencing the same issue. I am using a Samsung S20FE with Android 13. The tag is an ISO 14443-3A NXP - Mifare Classic 1k
> 
> I am just trying to read the tag, because I need the serialnumber from it. I am using the provided example code in this repo and the latest nfc kit version as of writing this. After I turned on all exceptions this is the exception I get. ![grafik](https://private-user-images.githubusercontent.com/49280032/284907798-8df4eda7-cc92-49e9-b512-5bf25380b67b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEyMDc0OTIsIm5iZiI6MTcwMTIwNzE5MiwicGF0aCI6Ii80OTI4MDAzMi8yODQ5MDc3OTgtOGRmNGVkYTctY2M5Mi00OWU5LWI1MTItNWJmMjUzODBiNjdiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI4VDIxMzMxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc4MDA1NjhlMGFlNzM0OGJjY2E2MWFiYWVjOGVhY2UxNjAwMGM4NTc1ZmVmNTY3ZjZmY2M5NGFlZDBjOThjNmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bGt_H1APvvhfJvE4k3Eb4qMdo5FIUbXO2dbBELl6RAM)

I also got that error. Error is here https://github.com/nfcim/flutter_nfc_kit/blob/master/lib/flutter_nfc_kit.g.dart#L47

More details:
```
#0      _$NFCTagFromJson (package:flutter_nfc_kit/flutter_nfc_kit.g.dart:47:52)
#1      new NFCTag.fromJson (package:flutter_nfc_kit/flutter_nfc_kit.dart:145:57)
#2      FlutterNfcKit.poll (package:flutter_nfc_kit/flutter_nfc_kit.dart:344:19)
<asynchronous suspension>
#3      _MyAppState.build.<anonymous closure> (package:betapper/main.dart:103:48)
<asynchronous suspension>
```
[image]

Areion0, 2023-12-05:
Same issue here, updated from version 3.3.1 to 3.4.1 and getting this when tag is picked up:

```
Unhandled Exception: type 'String' is not a subtype of type 'Map<String, dynamic>' in type cast
 #0      _$NFCTagFromJson (package:flutter_nfc_kit/flutter_nfc_kit.g.dart:47:52)
 #1      new NFCTag.fromJson (package:flutter_nfc_kit/flutter_nfc_kit.dart:145:57)
 #2      FlutterNfcKit.poll (package:flutter_nfc_kit/flutter_nfc_kit.dart:344:19)
```
 
Tag type: ISO 14443-3A
Device: Pixel 7 - Android 14

Harry-Chen, 2024-04-04:
This is fixed in v3.4.2.

closed 2024-04-04 by Harry-Chen