nfcim/flutter_nfc_kit #123: Error: The method 'toHexString' isn't defined for the class 'Uint8List'

luckycreationsindia, 2023-10-26:
Hi,

I'm getting following error while building the app on latest version of flutter:

`../../../.pub-cache/hosted/pub.dev/flutter_nfc_kit-3.3.3/lib/flutter_nfc_kit.dart:146:30: Error: The method 'toHexString' isn't defined for the class 'Uint8List'.
 - 'Uint8List' is from 'dart:typed_data'.
Try correcting the name to the name of an existing method, or defining a method named 'toHexString'.
    return NDEFRawRecord(id?.toHexString() ?? '', payload?.toHexString() ?? '',
                             ^^^^^^^^^^^
../../../.pub-cache/hosted/pub.dev/flutter_nfc_kit-3.3.3/lib/flutter_nfc_kit.dart:146:60: Error: The method 'toHexString' isn't defined for the class 'Uint8List'.
 - 'Uint8List' is from 'dart:typed_data'.
Try correcting the name to the name of an existing method, or defining a method named 'toHexString'.
    return NDEFRawRecord(id?.toHexString() ?? '', payload?.toHexString() ?? '',
                                                           ^^^^^^^^^^^
../../../.pub-cache/hosted/pub.dev/flutter_nfc_kit-3.3.3/lib/flutter_nfc_kit.dart:147:15: Error: The method 'toHexString' isn't defined for the class 'Uint8List'.
 - 'Uint8List' is from 'dart:typed_data'.
Try correcting the name to the name of an existing method, or defining a method named 'toHexString'.
        type?.toHexString() ?? '', this.tnf);
              ^^^^^^^^^^^
../../../.pub-cache/hosted/pub.dev/flutter_nfc_kit-3.3.3/lib/flutter_nfc_kit.dart:154:38: Error: The method 'toBytes' isn't defined for the class 'String'.
Try correcting the name to the name of an existing method, or defining a method named 'toBytes'.
        raw.typeNameFormat, raw.type.toBytes(), raw.payload.toBytes(),
                                     ^^^^^^^
../../../.pub-cache/hosted/pub.dev/flutter_nfc_kit-3.3.3/lib/flutter_nfc_kit.dart:154:61: Error: The method 'toBytes' isn't defined for the class 'String'.
Try correcting the name to the name of an existing method, or defining a method named 'toBytes'.
        raw.typeNameFormat, raw.type.toBytes(), raw.payload.toBytes(),
                                                            ^^^^^^^
../../../.pub-cache/hosted/pub.dev/flutter_nfc_kit-3.3.3/lib/flutter_nfc_kit.dart:155:58: Error: The method 'toBytes' isn't defined for the class 'String'.
Try correcting the name to the name of an existing method, or defining a method named 'toBytes'.
        id: raw.identifier == "" ? null : raw.identifier.toBytes());
                                                         ^^^^^^^
Target kernel_snapshot failed: Exception`

Does anyone else getting this issue?

There is not issue building this app without this package

fjvjim, 2023-10-27:
Hi, i have same problème in the version 3.3.3

CodeBurnCorp, 2023-10-27:
Hi. I have a same problem with 3.3.3 version and flutter 3.13.9

cliffordh, 2023-10-27:
Downgraded to 3.3.2 and same issue:

Error (Xcode): ../../../.pub-cache/hosted/pub.dev/flutter_nfc_kit-3.3.2/lib/flutter_nfc_kit.dart:146:30: Error: The method 'toHexString' isn't defined for the class 'Uint8List'.

cliffordh, 2023-10-27:
TEMP FIX... there was a change in ndef 0.3.3 which broke things... use this in your pubspec.yaml for now:

  flutter_nfc_kit: 3.3.2
  ndef: 0.3.2

luckycreationsindia, 2023-10-27:
Ok here's how I fixed it. I really don't know how it was working before. But here's quick fix,

just import following on flutter_nfc_kit.dart file inside flutter_nfc_kit package,

import 'package:ndef/utilities.dart';

This will fix the error.

Harry-Chen, 2023-10-29:
See https://github.com/nfcim/ndef/issues/20, thanks for all your reports. I will release a fixed version soon.

Harry-Chen, 2023-10-30:
Fixed by #124

closed 2023-10-30 by Harry-Chen