nfcim/flutter_nfc_kit #125: "Error: Type 'Uint8' not found" in version 3.4

mauriziopinotti, 2023-10-31:
Hi,

versions 3.4.0 and 3.4.1 seem to be broken:
```
$ flutter pub get && flutter run -d chrome
Launching lib/main.dart on Chrome in debug mode...
../lib/flutter_nfc_kit.dart:3:8: Error: Dart library 'dart:ffi' is not available on this platform.
import 'dart:ffi' show Uint8;
       ^
Context: The unavailable library 'dart:ffi' is imported through these packages:

    web_entrypoint.dart => package:flutter_nfc_kit_example => package:flutter_nfc_kit => dart:ffi

Detailed import paths for (some of) the these imports:

    web_entrypoint.dart => package:flutter_nfc_kit_example/main.dart => package:flutter_nfc_kit/flutter_nfc_kit.dart => dart:ffi

../lib/flutter_nfc_kit.dart:217:3: Error: Type 'Uint8' not found.
  Uint8 encode() {
  ^^^^^
../lib/flutter_nfc_kit.dart:257:40: Error: Type 'Uint8' not found.
  factory Iso15693RequestFlags.fromRaw(Uint8 raw) {
                                       ^^^^^
../lib/flutter_nfc_kit.dart:243:22: Error: 'Uint8' isn't a type.
    return result as Uint8;
                     ^^^^^
../lib/flutter_nfc_kit.dart:257:40: Error: 'Uint8' isn't a type.
  factory Iso15693RequestFlags.fromRaw(Uint8 raw) {
                                       ^^^^^
Waiting for connection from debug service on Chrome...             20.7s
Failed to compile application.
```

Version 3.3 was working fine.

jiegec, 2023-11-01:
We did not anticipate user to use the library from the web, interesting!

Harry-Chen, 2023-11-01:
I have pushed a fix in b47a7aa and will include it in the next release. Since the functionality on web platform has not changed in recent releases, you can still use old version (such as 3.3.3).

I'm also curious about how you use this library in browsers. Do you implement our WebUSB protocol on some hardware devices?

mauriziopinotti, 2023-11-01:
> I have pushed a fix in [b47a7aa]
Thanks, I confirm that works.

> I'm also curious about how you use this library in browsers. Do you implement our WebUSB protocol on some hardware devices?
Actually, I don't use is on Web, but the code base is shared between Web and Mobile.

closed 2023-11-01 by mauriziopinotti