nfcim/flutter_nfc_kit #126: Unhandled Exception: type 'String' is not a subtype of type 'Map<String, dynamic>' in type cast

adiwirya, 2023-11-01:
i got this error when taping my card to nfc reader

E/flutter ( 7380): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: type 'String' is not a subtype of type 'Map<String, dynamic>' in type cast
E/flutter ( 7380): #0      _$NFCTagFromJson
flutter_nfc_kit.g.dart:47
E/flutter ( 7380): #1      new NFCTag.fromJson
flutter_nfc_kit.dart:145
E/flutter ( 7380): #2      FlutterNfcKit.poll
flutter_nfc_kit.dart:344
E/flutter ( 7380): <asynchronous suspension>

this is the code 
     json['mifareInfo'] == null
          ? null
          : MifareInfo.fromJson(json['mifareInfo'] as Map<String, dynamic>),
    );

Harry-Chen, 2023-11-01:
Which platform are you on, Android or iOS?

Harry-Chen, 2023-11-01:
Please check whether cc5c098 fixes this problem.

martijn00, 2023-11-01:
@Harry-Chen you that fixes the problem. Thanks!

closed 2023-11-01 by Harry-Chen

martijn00, 2023-11-02:
Can you release a new version to pubdev?

Harry-Chen, 2023-12-06:
Now version 3.4.2 has been released. Sorry for the delay.

n0dz, 2023-12-13:
Hi there, on google pixel 6 it still opens the default nfc reader app I am using the 3.4.2 version of the plugin