nfcim/flutter_nfc_kit #127: Build fails with issue between java and kotlin

martijn00, 2023-11-01:
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_nfc_kit:compileDebugKotlin'.
> Inconsistent JVM-target compatibility detected for tasks 'compileDebugJavaWithJavac' (11) and 'compileDebugKotlin' (17).

Harry-Chen, 2023-11-01:
Maybe try higher JDK version like 17?

martijn00, 2023-11-01:
No that doesn't work and gives the same error. @mvanbeusekom you looked at this before. Can you give any hints?

Harry-Chen, 2023-11-01:
In flutter_nfc_kit, `sourceCompatibility` was set to 11:

https://github.com/nfcim/flutter_nfc_kit/blob/cc5c098700e618ba5629dbc33a00fce1cae12863/android/build.gradle#L22-L25

Would it be some other dependency that causes the problem?

martijn00, 2023-11-01:
Would it be possible to set it to VERSION_17?

Harry-Chen, 2023-11-01:
>  tasks 'compileDebugJavaWithJavac' (11) and 'compileDebugKotlin' (17).

Please note that the version 11 does not come from our project since we have no Java code at all. Maybe you should check other dependencies, or at least provide a minimal project that can reproduce this problem.

martijn00, 2023-11-01:
What about: https://github.com/flutter/packages/blob/main/packages/webview_flutter/webview_flutter_android/android/build.gradle#L26

And why not set it to VERSION_1_8 like the official Flutter plugins:
https://github.com/flutter/packages/blob/main/packages/webview_flutter/webview_flutter_android/android/build.gradle#L51

martijn00, 2023-11-01:
Here: https://github.com/nfcim/flutter_nfc_kit/blob/cc5c098700e618ba5629dbc33a00fce1cae12863/android/gradle/wrapper/gradle-wrapper.properties#L4

I use: https\://services.gradle.org/distributions/gradle-8.3-all.zip

AristideVB, 2024-03-13:
@Harry-Chen You should indeed set it to 17, see similar issue on other package https://github.com/matanshukry/flutter_google_places_sdk/issues/71

Harry-Chen, 2024-04-04:
Fixed in version 3.5.0.

closed 2024-04-04 by Harry-Chen

luckycreationsindia, 2024-07-19:
> Here:
> 
> https://github.com/nfcim/flutter_nfc_kit/blob/cc5c098700e618ba5629dbc33a00fce1cae12863/android/gradle/wrapper/gradle-wrapper.properties#L4
> 
> I use: https://services.gradle.org/distributions/gradle-8.3-all.zip

Facing this issue with latest flutter version with gradle 8+

Note:
I'm using gradle 8.7 due to compatibility issue with latest flutter version