nfcim/flutter_nfc_kit #129: Unable to read/write tags

fjvjim, 2023-11-03:
I use flutter nfc kit version: ^3.4.1 and it's not possible to read the tag. On version 3.3.3 works fine with sdk: '>=2.18.6 <3.0.0' , 'com.android.tools.build:gradle:7.3.1' and ext.kotlin_version = '1.9.0'

Harry-Chen, 2023-11-03:
The description is too vague. Maybe duplicated with #126 ?

fjvjim, 2023-11-03:
I couldn't check the bug because I tested on a real phone after the build.

fjvjim, 2023-11-03:
I just had null return on FlutterNfcKit.poll

Harry-Chen, 2023-11-03:
According to the logic on Android side, `poll` either succeeds with a Tag object, or fails with error. Are you sure you are getting `null`? What type of NFC tag are you trying to read?

fjvjim, 2023-11-03:
It's NfcA, MifareClassic, the same tag that I use with version 3.3.3 and I test that too with the nfc_manager library which works very well

Harry-Chen, 2023-11-03:
I still suspect the problem comes from #126. Did you try the `master` version?

fjvjim, 2023-11-06:
it's fine, thanks @Harry-Chen after use https://github.com/nfcim/flutter_nfc_kit/issues/126

closed 2023-11-06 by fjvjim