nfcim/flutter_nfc_kit #132: Build apk failed

bayramaliunuvar, 2023-11-20:
I'm still getting the same error.

**Error**
` What went wrong:
Execution failed for task :app:minifyReleaseWithR8.
> com.android.tools.r8.CompilationFailedException: Compilation failed to complete, origin: C:\Users\BAU\.gradle\caches\transforms-3\1d004bcf53e866d0bf97c62d164a539c\transformed\jetified-kotlin-reflect-1.9.10.jar:kotlin/reflect/jvm/internal/impl/serialization/deserialization/builtins/BuiltInsResourceLoader.class`

**Flutter doctor**
`[√] Flutter (Channel stable, 3.13.5, on Microsoft Windows [Version 10.0.22621.2715], locale tr-TR)
    • Flutter version 3.13.5 on channel stable at C:\src\flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 12fccda598 (9 weeks ago), 2023-09-19 13:56:11 -0700
    • Engine revision bd986c5ed2
    • Dart version 3.1.2
    • DevTools version 2.25.0

[√] Windows Version (Installed version of Windows is version 10 or higher)

[√] Android toolchain - develop for Android devices (Android SDK version 33.0.2)
    • Android SDK at C:\Users\BAU\AppData\Local\Android\Sdk
    • Platform android-34, build-tools 33.0.2
    • Java binary at: C:\Program Files\Android\Android Studio\jbr\bin\java
    • Java version OpenJDK Runtime Environment (build 17.0.6+0-b2043.56-9586694)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[√] Visual Studio - develop Windows apps (Visual Studio Community 2022 17.7.6)
    • Visual Studio at C:\Program Files\Microsoft Visual Studio\2022\Community
    • Visual Studio Community 2022 version 17.7.34221.43
    • Windows 10 SDK version 10.0.22000.0

[√] Android Studio (version 2022.2)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 17.0.6+0-b2043.56-9586694)

[√] VS Code (version 1.79.1)
    • VS Code at C:\Users\BAU\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.76.0

[√] Connected device (3 available)
    • Windows (desktop) • windows • windows-x64    • Microsoft Windows [Version 10.0.22621.2715]
    • Chrome (web)      • chrome  • web-javascript • Google Chrome 119.0.6045.160
    • Edge (web)        • edge    • web-javascript • Microsoft Edge 119.0.2151.72

[√] Network resources
    • All expected network resources are available.`

**pubspec.yaml**

`dependencies:
  flutter:
    sdk: flutter
  flutter_localizations:
    sdk: flutter

  # The following adds the Cupertino Icons font to your application.
  # Use with the CupertinoIcons class for iOS style icons.
  cupertino_icons: ^1.0.5
  font_awesome_flutter: ^10.3.0
  material_design_icons_flutter: ^7.0.7296
  http: ^1.0.0
  connectivity: ^3.0.6
  shared_preferences: ^2.0.17
  url_launcher: ^6.1.8
  # fluttertoast: 8.2.4
  xml: ^6.2.2
  turkish: ^0.2.0
  # nfc_manager: ^3.1.0
  flutter_nfc_kit: 3.4.1
  # flutter_nfc_reader: ^0.1.0
  # nfc_in_flutter: ^2.0.5
  # flutter_nfc_compatibility: ^0.0.4
  package_info: ^2.0.2
  firebase_messaging: ^14.2.1
  firebase_core: ^2.4.1
  google_mobile_ads: ^3.0.0
  flutter_credit_card: ^3.0.4
  animated_text_kit: ^4.2.2
  drawerbehavior: 2.3.1-dev.1
  launch_review: ^3.0.1
  # flutter_responsive_screen: ^1.0.0
  share: ^2.0.4

  open_settings: ^2.0.2
  flutter_speed_dial: ^6.2.0`

**app/build.gradle**

`plugins {
    id 'com.android.application'
}

def localProperties = new Properties()
def localPropertiesFile = rootProject.file('local.properties')
if (localPropertiesFile.exists()) {
    localPropertiesFile.withReader('UTF-8') { reader ->
        localProperties.load(reader)
    }
}

def flutterRoot = localProperties.getProperty('flutter.sdk')
if (flutterRoot == null) {
    throw new GradleException("Flutter SDK not found. Define location with flutter.sdk in the local.properties file.")
}

def flutterVersionCode = localProperties.getProperty('flutter.versionCode')
if (flutterVersionCode == null) {
    flutterVersionCode = '1'
}

def flutterVersionName = localProperties.getProperty('flutter.versionName')
if (flutterVersionName == null) {
    flutterVersionName = '1.0'
}

apply from: "$flutterRoot/packages/flutter_tools/gradle/flutter.gradle"

def keystorePropertiesFile = rootProject.file("key.properties")
def keystoreProperties = new Properties()
keystoreProperties.load(new FileInputStream(keystorePropertiesFile))

android {
    namespace "com.bauyazilim.xxx"
    compileSdkVersion 33
    ndkVersion flutter.ndkVersion

    compileOptions {
        sourceCompatibility JavaVersion.VERSION_11
        targetCompatibility JavaVersion.VERSION_11
    }

    sourceSets {
        main.java.srcDirs += 'src/main/kotlin'
    }

    defaultConfig {
        // TODO: Specify your own unique Application ID (https://developer.android.com/studio/build/application-id.html).
        
        // You can update the following values to match your application needs.
        // For more information, see: https://docs.flutter.dev/deployment/android#reviewing-the-gradle-build-configuration.
        minSdkVersion 19
        targetSdkVersion 33
        compileSdkVersion 33
        versionCode flutterVersionCode.toInteger()
        versionName flutterVersionName
        testInstrumentationRunner "androidx.test.runner.AndroidJUnitRunner"
    }

    signingConfigs{
        release{
            keyAlias keystoreProperties['keyAlias']
            keyPassword keystoreProperties['keyPassword']
            storeFile file(keystoreProperties['storeFile'])
            storePassword keystoreProperties['storePassword']
        }
    }
    buildTypes {
        release {
            ndk {
                abiFilters 'armeabi-v7a','arm64-v8a','x86_64'
            }
            // TODO: Add your own signing config for the release build.
            // Signing with the debug keys for now, so `flutter run --release` works.
            signingConfig signingConfigs.release
        }
    }
    
    lint {
        disable 'InvalidPackage'
    }
}

flutter {
    source '../..'
}

dependencies {
    implementation "org.jetbrains.kotlin:kotlin-stdlib-jdk8:$kotlin_version"
    implementation platform('com.google.firebase:firebase-bom:32.0.0')
    implementation 'com.google.firebase:firebase-analytics-ktx'

}`

**build.gradle**

`buildscript {
    ext.kotlin_version = '1.9.10'
}

plugins {
    id 'com.android.application' version '7.0.4' apply false
    id 'com.android.library' version '7.0.4' apply false
    id 'org.jetbrains.kotlin.android' version "$kotlin_version" apply false
}

allprojects {
    repositories {
        gradlePluginPortal()
        google()
        mavenCentral()
    }
}

rootProject.buildDir = '../build'
subprojects {
    project.buildDir = "${rootProject.buildDir}/${project.name}"
}
subprojects {
    project.evaluationDependsOn(':app')
}

tasks.register("clean", Delete) {
    delete rootProject.buildDir
}`

maxdiable, 2024-01-04:
i have the same problem :(
any solution ??

Harry-Chen, 2024-04-04:
Please track it in #145

closed 2024-04-04 by Harry-Chen