nfcim/flutter_nfc_kit #134: WIndows Platform

marwenbk, 2023-11-27:
Can you add support for windows ?

Harry-Chen, 2023-11-28:
Thanks for your feature request. Windows has PC/SC APIs to read smartcards, and [pcsc-lite](https://pcsclite.apdu.fr/) provides similar APIs on many more UNIX-like systems. So it is *possible* to add support on desktop platforms, but we have *no guarantee* on it since all maintainers are quite busy.

Harry-Chen, 2023-12-06:
I have done some very initial work on the [desktop](https://github.com/nfcim/flutter_nfc_kit/tree/desktop) branch. The core idea is that all platforms forward the call from `MethodChannel` (from Flutter) to a unified interface implemented by the `pcsc-interface` module, which then uses either `pcsc-lite` on Linux / macOS or native PCSC APIs on Windows.

Ideally the code in `pcsc-interface` should compile on all platforms, except for some compiler guards to handle some subtle differences.

dangfan, 2024-01-16:
Hi @marwenbk , I just found a library here: https://github.com/jjanku/dart_pcsc . That may help.

dangfan, 2024-05-31:
Hi @marwenbk, if you're looking for an implementation of CCID protocol on both mobile platforms and desktop platforms, try https://github.com/nfcim/ccid 
For now, we don't see a proper way to support desktop platforms as a smart card reading library.

closed 2024-05-31 by dangfan