nfcim/flutter_nfc_kit #137: FlutterNfcKit.poll won't throw exceptions when user cancelled on Android

joshuajylin, 2023-12-06:
Invoking FlutterNfcKit.finish to interrupt FlutterNfcKit.poll will throw exceptions on FlutterNfcKit.poll on iOS, but Android won't.

Harry-Chen, 2023-12-06:
What exception do you get? Is it in one of these below?

https://github.com/nfcim/flutter_nfc_kit/blob/cc5c098700e618ba5629dbc33a00fce1cae12863/ios/Classes/SwiftFlutterNfcKitPlugin.swift#L429-L450

joshuajylin, 2023-12-06:
> What exception do you get? Is it in one of these below?
> 
> https://github.com/nfcim/flutter_nfc_kit/blob/cc5c098700e618ba5629dbc33a00fce1cae12863/ios/Classes/SwiftFlutterNfcKitPlugin.swift#L429-L450

I got 409 SessionCanceled on iOS.
But nothing on Android, FlutterNfcKit.poll wasn't returned.

Harry-Chen, 2023-12-06:
> > What exception do you get? Is it in one of these below?
> > https://github.com/nfcim/flutter_nfc_kit/blob/cc5c098700e618ba5629dbc33a00fce1cae12863/ios/Classes/SwiftFlutterNfcKitPlugin.swift#L429-L450
> 
> I got 409 SessionCanceled on iOS. But nothing on Android, FlutterNfcKit.poll wasn't returned.

Both are currently expected behaviors. Terminating a session on iOS causes a callback with `Error`, while on Android there is no such thing, only that the polling timer task is canceled.

I tend not to change them because it might break existing users.

closed 2024-01-21 by dangfan