nfcim/flutter_nfc_kit #146: Sunmi Device P2 not support

Kimsoer, 2024-01-16:
Dear team, Thank you so much for your great library, I have issue to use it with sunmi device p2 and p2 pro, I try to use it for checking nfc but it response not support, Is it possible to use it with sunmi device, thank you

dangfan, 2024-01-16:
Thank you for your feedback and for using our library. It's important to note that these devices do not use the standard Android NFC interface. As a result, they require specific development based on the manufacturer's documentation. Unfortunately, this means that our library does not support these devices due to their unique implementation.

closed 2024-01-16 by dangfan

Kimsoer, 2024-01-16:
Oh , I got it thank you so much @dangfan