nfcim/flutter_nfc_kit #150: ISO 15693 Transceive error

AppsService, 2024-02-15:
```
[NFCTagReaderSession transceive:tagUpdate:error:]:888 Error Domain=NFCError Code=102 "Stack Error" UserInfo={NSLocalizedDescription=Stack Error, NSUnderlyingError=0x28141c990 {Error Domain=nfcd Code=10 "Invalid Parameter" UserInfo={NSLocalizedDescription=Invalid Parameter}}}
```

Transceive error on iOS but working on android.

rostopira, 2024-02-22:
Doesn't seem like a library issue to me.
I'm using it without any issues with NFC-V.
Could you provide a code example? Maybe I can help

AppsService, 2024-02-23:
Here is code example

```
 Uint8List command = Uint8List(12);
 command[0] = 0x60; // Flag address
 command[1] = 0x23; // Read multiple blocks
 command[2...9] = UID
 command[10] = 0x00; // Start block
 command[11] = 0x04; // End block
 Uint8List result = await FlutterNfcKit.transceive(command);
```

Thank you

rostopira, 2024-02-23:
Underlying implementation for iOS is currently intended for custom commands.
However, I'm not sure, that library is an issue.
Can you try to play with request flags?
This will help
```
/**
    From iOS headers
    NFCISO15693RequestFlag.address: NFCISO15693RequestFlag(rawValue: 32)
    NFCISO15693RequestFlag.commandSpecificBit8: NFCISO15693RequestFlag(rawValue: 128)
    NFCISO15693RequestFlag.dualSubCarriers: NFCISO15693RequestFlag(rawValue: 1)
    NFCISO15693RequestFlag.highDataRate: NFCISO15693RequestFlag(rawValue: 2)
    NFCISO15693RequestFlag.option: NFCISO15693RequestFlag(rawValue: 64)
    NFCISO15693RequestFlag.protocolExtension: NFCISO15693RequestFlag(rawValue: 8)
    NFCISO15693RequestFlag.select: NFCISO15693RequestFlag(rawValue: 16)
 * **/

abstract class NfcVFlag {
  static const dualSubCarriers = 1;
  static const highDataRate = 2;
  static const protocolExtension = 8;
  static const select = 16;
  static const address = 32;
  static const option = 64;
  static const commandSpecificBit8 = 128;
}
```
Currently you are using `address | option`.
Try using `highDataRate | address` or just `address`.

If that won't help you, I can add [readMultipleBlocks](https://developer.apple.com/documentation/corenfc/nfciso15693tag/3043809-readmultipleblocks) and [fastReadMultipleBlocks](https://developer.apple.com/documentation/corenfc/nfciso15693tag/3778593-fastreadmultipleblocks) to the library, that should work.

rostopira, 2024-02-23:
Alternatively, you can try using this library: https://github.com/okadan/flutter-nfc-manager
It has support for all platform commands exposed in Dart: https://github.com/okadan/flutter-nfc-manager/blob/master/lib/src/platform_tags/iso15693.dart

AppsService, 2024-02-26:
Error shown for custom command even thought I use flutter-nfc-manager. For now I'll use flutter-nfc-manager for iOS but still same error occured when comes to custom command. Thanks for your suggestions and also maybe you can add those readMultipleBlock and fastReadMultipleBlock in your library and also maybe add more command based on nfc tag.

Harry-Chen, 2024-02-26:
> Error shown for custom command even thought I use flutter-nfc-manager. For now I'll use flutter-nfc-manager for iOS but still same error occured when comes to custom command. Thanks for your [...]

Maybe it's not related to one specific library, but the command and/or your tag.

AppsService, 2024-02-26:
> Maybe it's not related to one specific library, but the command and/or your tag.

Yes it is. The command error on several try and error. Nothing to do with the library. But if the library can add on several function without using custom command it will be great.

Harry-Chen, 2024-02-26:
> Yes it is. The command error on several try and error. Nothing to do with the library. But if the library can add on several function without using custom command it will be great.

Adding one specific API is easy, as `flutter-nfc-manager` does. But we try to be "real" cross-platform, that is, not simply adding API wrappers from different platforms, but adding one more layer of abstraction and hiding the difference to developers. Thus we would be cautious on new APIs.

But sure, we will consider your request and try to include more platform-provided APIs instead of having users to use raw / custom commands.

closed 2024-02-27 by AppsService