nfcim/flutter_nfc_kit #156: iPhone 15 pro |  NFCAvailability.not_supported

RobertBrunhage, 2024-04-18:
Polling returns 

```
Uncaught exception: PlatformException(500, Generic NFC Error, The operation couldn’t be completed. (NFCError error 4099.), null)
#0 StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:651:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
<asynchronous suspension>
```

and availablility returns `NFC availability: NFCAvailability.not_supported`

Harry-Chen, 2024-04-18:
Please check:

* Have you added entitlements as <https://github.com/nfcim/flutter_nfc_kit?tab=readme-ov-file#ios>? 
* Can you reproduce this problem consistently (if so, please provide minimal code).
* Try rebooting the phone, as there are indeed some strange issues inside CoreNFC that we could not deal with.

RobertBrunhage, 2024-04-25:
Sorry for the late reply, found the issue. XCode didn't update the entitlements and had to manually add the entries.

Thanks for the speedy response!

closed 2024-04-25 by RobertBrunhage