nfcim/flutter_nfc_kit #158: Project requires a newer version of the Kotlin Gradle plugin.

markHamilton34, 2024-05-08:
Hello,

I'm trying to run my application, however, I can't because it asks me to update the Kotlin version. I configured the build.gradle file as follows but it still gives the same error:

```
buildscript { 
    ext.kotlin_version = '1.9.24' 
}

plugins {
  ...
}
```

I have also tried with other earlier versions of Kotlin but it still doesn't work. Some of the errors that appear are as follows:

```
e:/Users/user/.pub-cache/hosted/pub.dev/flutter_nfc_kit-3.4.2/android/src/main/kotlin/im/nfc/flutter_nfc_kit/MifareUtils.kt: (105, 17): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/user/.gradle/caches/transforms-3/f63a9622c8bb10f864c02e086298cb3a/transformed/jetified-kotlin-stdlib-1.9.10.jar!/kotlin/Unit.class
e: /Users/user/.pub-cache/hosted/pub.dev/flutter_nfc_kit-3.4.2/android/src/main/kotlin/im/nfc/flutter_nfc_kit/MifareUtils.kt: (106, 24): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
```
I tried to set the version to 1.7.1 but it still doesn't work.
How can I solve it? 

Thank you for your assistance.

Harry-Chen, 2024-05-08:
Seems your project involves some dependencies that requires older version of kotlin. Please check your global config & third-party libraries.

markHamilton34, 2024-05-08:
When I run the application without the flutter_nfc_kit package, it works correctly. However, when I install the package in the application and run it, those errors appear and I cannot run the application.

adammeghji, 2024-05-08:
I'm seeing this error as well on a fresh project where the only dependency is `flutter_nfc_kit`, so appears to be a regression.

markHamilton34, 2024-05-14:
I solved the problem.

I was changing the `ext.kotlin_version` in the file `flutter_project/android/app/build.gradle`. But it should be changed in the file `flutter_project/android/build.gradle` in the buildscript:

```
buildscript {
     ext.kotlin_version = '1.9.0'
     ...
}
```

I'm commenting in case someone else has the same issue.

closed 2024-05-16 by markHamilton34