nfcim/flutter_nfc_kit #159: How to create MifareClassic instance after poll

markHamilton34, 2024-05-16:
Hello,

I am trying to create an instance of MifareClassic object after getting the NFCTag tag using FlutterNfcKit.poll. However, I cannot convert NFCTag to NfcTag which is required to create a MifareClassic instance as far as I understand.

Is it possible to create a MifareClassic instance after getting the tag of the NFC Card using FlutterNfcKit.poll?

Thank you!

Harry-Chen, 2024-05-17:
Sorry, but I cannot understand your problem. The [MifareInfo](https://pub.dev/documentation/flutter_nfc_kit/latest/flutter_nfc_kit/MifareInfo-class.html) in our library is only metadata. You should always operate on the tag with `FlutterNfcKit.foo` functions.

Can you provide some code examples?

markHamilton34, 2024-05-17:
I could operate with `FlutterNfcKit.foo` functions. I mean I was able to understand the doubt I had about MifareClassic.

But I have another problem: 
When I try to read an NFC card, I get the following error:
`Failed assertion: line 452 pos 12: 'T is String || T is Uint8List': is not true.`

I am using version 3.4.2 (due to some problems with versions with other dependencies so I cannot upgrade flutter_kit_nfc to a higher version) but in the issue #148, someone had the same problem, and in version 3.5.0 this validation was changed, but I still get the same error (even with that validation because I modified it to try if that worked). 

I am sending an object of type Uint8List in this way:

```
Uint8List array = Uint8List.fromList([0xA1, 0xA1, 0xA1, 0xA1, 0xA1, 0xA1]);
await FlutterNfcKit.authenticateSector(1, keyA: keyArray);
```

How could I solve it?

Harry-Chen, 2024-05-20:
> I could operate with `FlutterNfcKit.foo` functions. I mean I was able to understand the doubt I had about MifareClassic.
> 
> But I have another problem: When I try to read an NFC card, I get the following error: `Failed assertion: line 452 pos 12: 'T is String || T is Uint8List': is not true.`
> 
> I am using version 3.4.2 (due to some problems with versions with other dependencies so I cannot upgrade flutter_kit_nfc to a higher version) but in the issue #148, someone had the same problem, and in a higher version this validation was changed, but I still get the same error (even with that validation because I modified it to try if that worked).
> 
> I am sending an object of type Uint8List in this way:
> 
> ```
> Uint8List keyArray = Uint8List.fromList([0xA1, 0xA1, 0xA1, 0xA1, 0xA1, 0xA1]);
> await FlutterNfcKit.authenticateSector(1, keyA: keyArray);
> ```
> 
> How could I solve it?

If your dependency problem is related to `js`, you can try using

https://github.com/nfcim/nfsee/blob/1302225df144a2747ecc8130bf43edee2440f1a4/pubspec.yaml#L27-L28

I think the problem described in #148 is fixed in 3.5.0.

Harry-Chen, 2024-05-20:
> Now I can run my app using 3.5.0 version thanks to adding `dependency_overrides: js: "^0.6.4"` as you recommended, thank you!
> 
> The problem now is that I have the same error as before in this function of the package:
> 
> ```
> static Future<bool> authenticateSector<T>(int index,
>       {T? keyA, T? keyB}) async { 
>      assert((keyA.runtimeType == String || keyA.runtimeType == Uint8List) ||
>         (keyB.runtimeType == String || keyB.runtimeType == Uint8List));
>       ....
> }
> ```
> 
> Error:
> 
> ```
> I/flutter (29257): ERROR:'package:flutter_nfc_kit/flutter_nfc_kit.dart': Failed assertion: line 459 pos 12: '(keyA.runtimeType == String || keyA.runtimeType == Uint8List) ||
> I/flutter (29257):         (keyB.runtimeType == String || keyB.runtimeType == Uint8List)': is not true.
> ```
> 
> As before, I am sending an object of type Uint8List like this:
> 
> ```
> Uint8List keyArray = Uint8List.fromList([0xA1, 0xA1, 0xA1, 0xA1, 0xA1, 0xA1]);
> await FlutterNfcKit.authenticateSector(1, keyA: keyArray);
> ```
> 
> I don't know why it returns false in that condition. How could I solve it?

The following code prints `true` on <dartpad.dev> but `false` on native platforms. It might be some behavioural differences or Dart language......

I will try to release a fix soon.

```dart
import 'dart:typed_data';

bool authenticateSector<T>(int index,
      {T? keyA, T? keyB}) { 
     return ((keyA.runtimeType == String || keyA.runtimeType == Uint8List) ||
        (keyB.runtimeType == String || keyB.runtimeType == Uint8List));
}

void main() {
  Uint8List keyArray = Uint8List.fromList([0xA1, 0xA1, 0xA1, 0xA1, 0xA1, 0xA1]);
  print(authenticateSector(0, keyA: keyArray));
}
```

markHamilton34, 2024-05-20:
Thank you!

I am not sure if it's because I edited the assert function but I tried to put it in true like this `assert(true);` but now when I try to read an NFC card I get this error and the app closes:

```
E/AndroidRuntime( 3988): FATAL EXCEPTION: Thread-8
E/AndroidRuntime( 3988): Process: com.company.example, PID: 3988
E/AndroidRuntime( 3988): java.lang.IllegalStateException: Call connect() first!
E/AndroidRuntime( 3988): 	at android.nfc.tech.BasicTagTechnology.checkConnected(BasicTagTechnology.java:52)
E/AndroidRuntime( 3988): 	at android.nfc.tech.MifareClassic.authenticate(MifareClassic.java:371)
E/AndroidRuntime( 3988): 	at android.nfc.tech.MifareClassic.authenticateSectorWithKeyA(MifareClassic.java:339)
E/AndroidRuntime( 3988): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$7.invoke(FlutterNfcKitPlugin.kt:324)
E/AndroidRuntime( 3988): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$7.invoke(FlutterNfcKitPlugin.kt:318)
E/AndroidRuntime( 3988): 	at kotlin.concurrent.ThreadsKt$thread$thread$1.run(Thread.kt:30)
I/Process ( 3988): Sending signal. PID: 3988 SIG: 9
```

I believe it is because I force the assert to be true and maybe the return gets that error because of the principal problem of the assertion as you mentioned.

Harry-Chen, 2024-05-20:
> authenticateSector

Thanks for pointing this out. I indeed forgot to call `connect` in native Android plugin. I will fix that soon.

Harry-Chen, 2024-05-20:
@markHamilton34 Please try the master branch of this repo when available. If that works, I will release to dart pub.

markHamilton34, 2024-05-20:
Now it works! 
I tried the master branch of this repo and authenticateSector works fine. Thank you for your support!

I have another problem related to reading blocks that probably is something I'm missing.
I'm reading the first block of an NFC card this way:

```
for (int i = sector! * 4; i < (sector * 4) + blocks.length; i++) {
        try {
          print(i);
          Uint8List? info = await FlutterNfcKit.readBlock(i);
          print('correct: ' + i.toString());
        } catch (e) {
          print('ERROR READING BLOCK $i: ' + e.toString());
        }
      }
```
But I get this error:

```
I/flutter (18388): 8
E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): Read block error
E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): java.io.IOException: Transceive failed
E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at android.nfc.TransceiveResult.getResponseOrThrow(TransceiveResult.java:52)
E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at android.nfc.tech.BasicTagTechnology.transceive(BasicTagTechnology.java:154)
E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at android.nfc.tech.MifareClassic.readBlock(MifareClassic.java:425)
E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at im.nfc.flutter_nfc_kit.MifareUtils.readBlock(MifareUtils.kt:85)
E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$8.invoke(FlutterNfcKitPlugin.kt:356)
E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$8.invoke(FlutterNfcKitPlugin.kt:353)
E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at kotlin.concurrent.ThreadsKt$thread$thread$1.run(Thread.kt:30)
I/flutter (18388): ERROR READING BLOCK  8: PlatformException(500, Communication error, Transceive failed, null)
```

From 1st to 7th block works perfect, but from 8th to the last block (11) throws that error.
I tried to fix it but couldn't.

Harry-Chen, 2024-05-20:
> Now it works! I tried the master branch of this repo and authenticateSector works fine. Thank you for your support!
> 
> I have another problem related to reading blocks that probably is something I'm missing. I'm reading the first block of an NFC card this way:
> 
> ```
> for (int i = sector! * 4; i < (sector * 4) + blocks.length; i++) {
>         try {
>           print(i);
>           Uint8List? info = await FlutterNfcKit.readBlock(i);
>           print('correct: ' + i.toString());
>         } catch (e) {
>           print('ERROR READING BLOCK $i: ' + e.toString());
>         }
>       }
> ```
> 
> But I get this error:
> 
> ```
> I/flutter (18388): 8
> E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): Read block error
> E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): java.io.IOException: Transceive failed
> E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at android.nfc.TransceiveResult.getResponseOrThrow(TransceiveResult.java:52)
> E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at android.nfc.tech.BasicTagTechnology.transceive(BasicTagTechnology.java:154)
> E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at android.nfc.tech.MifareClassic.readBlock(MifareClassic.java:425)
> E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at im.nfc.flutter_nfc_kit.MifareUtils.readBlock(MifareUtils.kt:85)
> E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$8.invoke(FlutterNfcKitPlugin.kt:356)
> E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$8.invoke(FlutterNfcKitPlugin.kt:353)
> E/im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin(18388): 	at kotlin.concurrent.ThreadsKt$thread$thread$1.run(Thread.kt:30)
> I/flutter (18388): ERROR READING BLOCK  8: PlatformException(500, Communication error, Transceive failed, null)
> ```
> 
> From 4th to 7th block works perfect, but from 8th to the last block (11) throws that error. I tried to fix it but couldn't.

What `MifareInfo` did you get from your tag? Have you tried `readSector` on the same blocks?

markHamilton34, 2024-05-20:
I solved the problem, it was something I was doing incorrectly with my the logic of my code.

Now, the first time I read the NFC card works perfect, reads every block and sector and gives the correct information. The problem is that it only works the first time I read the NFC. If I read it again the app crashes and gives the following error:

```
E/AndroidRuntime(26838): FATAL EXCEPTION: Thread-57
E/AndroidRuntime(26838): Process: com.company.example, PID: 26838
E/AndroidRuntime(26838): java.lang.SecurityException: Permission Denial: Tag ( ID: 83 ED D7 1D ) is out of date
E/AndroidRuntime(26838): 	at android.nfc.Tag.getTagService(Tag.java:388)
E/AndroidRuntime(26838): 	at android.nfc.tech.BasicTagTechnology.isConnected(BasicTagTechnology.java:63)
E/AndroidRuntime(26838): 	at android.nfc.tech.MifareClassic.isConnected(MifareClassic.java:73)
E/AndroidRuntime(26838): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$switchTechnology$1.invoke(FlutterNfcKitPlugin.kt:93)
E/AndroidRuntime(26838): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$switchTechnology$1.invoke(FlutterNfcKitPlugin.kt:92)
E/AndroidRuntime(26838): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$7.invoke(FlutterNfcKitPlugin.kt:321)
E/AndroidRuntime(26838): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$handleMethodCall$7.invoke(FlutterNfcKitPlugin.kt:318)
E/AndroidRuntime(26838): 	at kotlin.concurrent.ThreadsKt$thread$thread$1.run(Thread.kt:30)
I/Process (26838): Sending signal. PID: 26838 SIG: 9
```

I'm using `await FlutterNfcKit.poll();` to read the NFC card. I also tried a longer Duration in the parameter but the app still crashes.

Is it something that I'm missing when reading the NFC Card?

Harry-Chen, 2024-05-21:
v3.5.1 is released.

closed 2024-05-21 by Harry-Chen