nfcim/flutter_nfc_kit #164: Failed assertion: T is String || T is Uint8List: is not ture

Saintqqq, 2024-05-27:
hi, 
I used the sample code and tried to write data into a specific block, but an error occurred.

message:  flutter_nfc_kit.dart: failed assertion: line 496 pos 12: T is String|| T is Uint8List: is not ture

this is sample code 
```
await FlutterNfcKit.writeBlock(
    1, // index
    [0xde, 0xad, 0xbe, 0xff], // data
    iso15693Flags:
        Iso15693RequestFlags(), // optional flags for ISO 15693
    iso15693ExtendedMode: false // use extended mode for ISO 15693
);
```
Does anyone have any suggestions？  Thanks

Harry-Chen, 2024-05-27:
Fix to this issue will be released in the next version.

Saintqqq, 2024-05-28:
> Fix to this issue will be released in the next version.

Glad to hear that, and look forward to your updates soon. Thank you.

Saintqqq, 2024-05-28:
hi. 
When will you have a new version? 
Or are there other ways I can try to adjust first, for example, transceive?

thanks

Saintqqq, 2024-05-29:
> v3.5.2 is released.

hi Harry
There are still errors

this is my code which is copy from sample

```
await FlutterNfcKit.writeBlock(
    15, // index
    [0xde, 0xad, 0xbe, 0xff], // data
    iso15693RequestFlag: Iso15693RequestFlag(), // optional flags for ISO 15693
    iso15693ExtendedMode: false // use extended mode for ISO 15693
  );
```
this error message is 
flutter_nfc_kit.dart: failed assertion: line 492 pos 12: data is String || data is Uint8List: is not true.

can u update the sample about "writeBlock" ? or  FlutterNfcKit.transceive. sample 
Thanks

Saintqqq, 2024-05-29:
finnally,  this code is working.

```
 Uint8List data = Uint8List.fromList([0x12, 0x34, 0x56, 0x78]);
  await FlutterNfcKit.writeBlock(
      16, // indexdata
      data, // data
      iso15693Flags:
          Iso15693RequestFlags(), // optional flags for ISO 15693
      iso15693ExtendedMode: true // use extended mode for ISO 15693
      );
```

thanks