nfcim/flutter_nfc_kit #171: Effect of Extending NFC Poll Timeout and Its Necessity

haizadvnet, 2024-06-04:
First, I want to thank the developers and contributors for creating/maintaining this package.

I am currently using the [timeout] parameter for the Future<NFCTag> poll function, which defaults to 20 seconds. I would like to poll data without specifying a timeout. What impact would extending the timeout to 1 minute or longer, or setting it to never expire, have on the NFC polling process? Additionally, why is it necessary to provide a timeout parameter?

Many thanks.

markHamilton34, 2024-06-07:
Hello,

I would like to know it too.
Thanks!

Harry-Chen, 2024-06-07:
> What impact would extending the timeout to 1 minute or longer, or having it never expire, have on the polling process?

For Android, we just set a timer to disable reader mode after the given timeout value. So it depends on the system whether a longer timeout has different behavior, and I have little experience on this. If Android supports longer (or unlimited) polling time, it is safe to pass any value to `poll`. You can also create a PR to enable special values (e.g. 0 for "never").

> Additionally, why is it necessary to provide a timeout parameter?

CoreNFC on iOS has set an internal time limit for any polling session (seemingly 1min), and `readerSessionInvalidationErrorSessionTimeout` will be thrown after the limit. Since this library is cross-platform, a default timeout is set for Android users, so the `poll` method behaves similarly on both platforms.

haizadvnet, 2024-06-08:
Thank you for your detailed response. I appreciate the suggestion to create a PR and will consider it for future improvements, though it may not be a priority since my focus is primarily on Android development.

closed 2024-06-08 by haizadvnet