nfcim/flutter_nfc_kit #172: IOS transceive NFCTagType.iso7816

HVLoc, 2024-06-11:
Here's the translation of your request into English:

---

I use the following code to read NFC tags as card IDs on iOS:

```dart
if (tag.type == NFCTagType.iso7816) {
  String result1 = await FlutterNfcKit.transceive("00a40400");
  setState(() {
    _result = '1: $result1\n';
  });
}
```

And I encounter an error:

```
[CoreNFC] -[NFCTagReaderSession setAlertMessage:]:101 (null)
[CoreNFC] -[NFCTagReaderSession transceive:tagUpdate:error:]:883 Error Domain=NFCError Code=2 "Missing required entitlement" UserInfo={NSLocalizedDescription=Missing required entitlement}
```

I have added the following to the `info.plist` file:

```xml
<key>com.apple.developer.nfc.readersession.felica.systemcodes</key>
<array>
  <string>12FC</string>
  <string>6A82</string>
  <string>8008</string>
</array>
<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
<array>
  <string>A0000002471001</string>
  <string>A0000002472001</string>
  <string>A000000151000000</string>
  <string>A00000000386980701</string>
  <string>A000000003101001</string>
  <string>A000000003101002</string>
  <string>A0000000041010</string>
  <string>A0000000042010</string>
  <string>A0000000044010</string>
  <string>44464D46412E44466172653234313031</string>
  <string>D2760000850100</string>
  <string>D2760000850101</string>
  <string>00000000000000</string>
</array>
```

And the `Runner.entitlements` file:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
  <key>com.apple.developer.nfc.readersession.formats</key>
  <array>
    <string>TAG</string>
    <string>NDEF</string>
  </array>
</dict>
</plist>
```

dangfan, 2024-06-11:
When you use the Select AID command (A4 with P1=04), you have to specify an AID and add it to the info.plist.

HVLoc, 2024-06-11:
> When you use the Select AID command (A4 with P1=04), you have to specify an AID and add it to the info.plist.
Here's the translation of your request into English:

---

I have added the AID, but it seems to be incomplete. Do you have any other AID sets? When using (A4 with P1=04), it works on Android.

dangfan, 2024-06-11:
Unfortunately, you have to specify a specific AID when running on iOS. For example, if your AID is A0000002471001, you need to send 00A4040007A0000002471001.

HVLoc, 2024-06-11:
> Unfortunately, you have to specify a specific AID when running on iOS. For example, if your AID is A0000002471001, you need to send 00A4040007A0000002471001.

Here's the translation of your request into English:

---

Thank you for the guidance, but how can I obtain a specific AID? I can only get the ID of the card, for example: 08C36DAC.

dangfan, 2024-06-11:
You can search for the corresponding AID on the Internet based on the card type.

HVLoc, 2024-06-11:
> Unfortunately, you have to specify a specific AID when running on iOS. For example, if your AID is A0000002471001, you need to send 00A4040007A0000002471001.

---

Thank you for the suggestion, I have managed to solve part of the issue, but I am encountering a problem with the READ_BINARY command. I have a model like this: `APDU(CLA:00 INS:B0 P1:9C P2:00 Le:8 Lc:2 Data:null)`. Do you have any solutions or search keywords for this? It is giving the error "Security status not satisfied".

Harry-Chen, 2024-06-11:
This is not an issue related to the library. Please refer to the manual of your tags for more details.

closed 2024-06-11 by Harry-Chen