nfcim/flutter_nfc_kit #174: new problem when trying to add this package

yaakoubDj, 2024-06-19:
when i try to add this package to my project i get a lot of errors the most about Kotlin version , when i remove this package and try again everything work !! , so the problem in this package , i hope fix it because i need this package to get id of NFC tag.

markHamilton34, 2024-06-25:
#158 This could help you.

dpsavaliya, 2024-07-01:
same issue. Is there any solution?

dpsavaliya, 2024-07-01:
> #158 This could help you.

Not working

Harry-Chen, 2024-07-01:
> when i remove this package and try again everything work !! , so the problem in this package

Not necessarily. Would you please provide a minimal reproducing code? E.g. the Kotlin and Flutter version you are using, other dependencies, and the specific errors you got.

dpsavaliya, 2024-07-01:
I have solved the issue by changing the kotlin version to `"1.9.23"` in `flutter_project/android/setting.gradle` file. 

`plugins {
    id "org.jetbrains.kotlin.android" version "1.9.23" apply false
}`

I'm commenting in case someone else has the same issue.

Harry-Chen, 2024-07-01:
The the problem is related to incompatible Kotlin version, mostly because the Kotlin gradle plugin you are using is too old. Good to see that you have solved it.

closed 2024-07-01 by Harry-Chen